exelban/stats #2182: No Icons or Widgets Display

sapph42, 2024-10-14:
MacBook Pro 16" M3 Pro, running public beta 15.1 beta 3. (I am aware betas are buyer beware, but I am willing to provide any needed assistance to dev to ensure Stats works properly with Sequoia)

Just installed HiddenBar and Stats.  On installation of Stats, Network Widget was briefly visible, but not any more. 

I have restarted the machine, and disabled HiddenBar, to no avail.  Before disabling it, I also checked that the Always Hidden feature was not enabled. I have checked Console for Crash Logs, and not found anything for the last several days.

Attached is a screenshot showing Stats in Activity Monitor, the Battery widget in Stats as enabled (among several other widgets), my About window, and of course, the Menu Bar with no Stats icons.

[image]

sapph42, 2024-10-14:
Further testing seems to show that the global application setting "Combined Modules" appears to be the general culprit.  Additionally, for any given module, selecting "Merge Widgets" appears to hide that module entirely.

exelban, 2024-10-14:
I have idea what the idea of this issue. But does not seems it's related to the Stats.

closed 2024-10-14 by exelban