exelban/stats #2183: Weird public IP suffix

zigapovhe, 2024-10-16:
**Describe the bug**
Today I've noticed that public IP has some weird suffix (X3). I know, this is not part of the IP address, but I'm interested what could it be?

[image]

**Details:**
 - Device: Macbook Pro M1
 - macOS: 15.0.1
 - Application version: 2.11.14

exelban, 2024-10-16:
hi. It's a contry code provided by macOS.

zigapovhe, 2024-10-16:
Huh, that's unusual. Why would I want to see my country code? Especially when MacOS is showing the wrong one? Can you remove it?

exelban, 2024-10-16:
there were a lot of asks about that, and also it's useful when using a VPN