exelban/stats #2194: Clicking preference "gear" does not refocus open preference window

fiveangle, 2024-10-25:
Clicking the preferences "gear" icon within a menu does not bring preferences window focus forward if it is already open. This causes confusion as it appears clicking the icon does nothing if the preference window is already open but underneath existing windows.

Desired behavior: clicking preferences "gear" icon should always bring preference window focus forward, even if preference window is already open.

https://github.com/user-attachments/assets/45a8176f-6448-4e89-bf42-5486bd08d334

fiveangle, 2024-10-25:
ps- I absolutely ♥️ Stats and tell all my friends about it. You are doing an amazing job !

exelban, 2024-10-25:
Hi. Since I cannot reproduce that bug please try this build:
[Stats.dmg.zip](https://github.com/user-attachments/files/17519983/Stats.dmg.zip)

fiveangle, 2024-11-01:
Same observation with this 2.11.14 version you posted above.

```
Hardware Overview:

  Model Name:	MacBook Pro
  Model Identifier:	MacBookPro11,4
  Processor Name:	Quad-Core Intel Core i7
  Processor Speed:	2.8 GHz
  Number of Processors:	1
  Total Number of Cores:	4
  L2 Cache (per Core):	256 KB
  L3 Cache:	6 MB
  Hyper-Threading Technology:	Enabled
  Memory:	16 GB
  System Firmware Version:	489.0.0.0.0
  OS Loader Version:	540.120.3~37
  SMC Version (system):	2.29f24

System Software Overview:

  System Version:	macOS 12.7.6 (21H1320)
  Kernel Version:	Darwin 21.6.0
  Boot Volume:	silicon
  Boot Mode:	Normal
  Secure Virtual Memory:	Enabled
  System Integrity Protection:	Enabled
  Time since boot:	28 days 7:59

```

exelban, 2024-11-03:
Sorry, but since that build does not solve the problem then the problem is not in Stats. I saw this behavior on any of my Macs and also the popup window works in the same way as the settings. So since the popup is showing and settings are not the problem somewhere else.

closed 2024-11-03 by exelban

fiveangle, 2024-11-04:
So you are saying that the problem is with macOS ? That is a pretty bold claim.

The issue certainly does appear to be within the control of Stats, even if Stats is "not the issue" as you state. When one clicks the menubar item to activate it, then clicks the preference icon, revealing the preference window, once you click the background Safari window to move the preference focus behind the Safari window, once you click the Status menu item again, if you then click off the memu, the prefs window focus comes forward. It is only when in that situation that you do not click off the menu, but instead click the Prefs gear icon again that it does not bring the behind Prefs window to forward focus.

Could the draw window routine be checking to see if Prefs window world already exists rather than exists and has forward focus ?

Even if it is "working as expected", it isn't. You could mimic the behavior to get the correct result by choosing to, in the event of clicking on the Prefs gear icon with the Prefs window already present, instead de-activate the Menu first (which brings the Prefs window forward) rather than going through the normal routine to launch the prefs window (which is already present but not in focus) ?

There must be some solution 🤷‍♂️

fiveangle, 2024-11-04:
See how the menu click behavior is different between clicking the menu and clicking the preference:

https://github.com/user-attachments/assets/dad5ef1b-a03d-4126-9d36-49c769779946

PuzzledUser, 2025-02-07:
Perhaps this is related to the fact that the current application window _loses_ focus when clicking a
Stats widget. Other widgets don't steel focus from the frontmost window.

fiveangle, 2025-02-10:
Possibly, but it very much is confusing. I have it on my todo list to revisit iOS/macOS development so crafting a fix for this is on my "todo", if only for helping me better re-acquaint myself with development on the platform. But I am confident that whatever the reason it is certainly something that can be solved without unapproved "hack" methods.