exelban/stats #2197: Request Reverting Sensor View to Previous Version

ukateki, 2024-10-27:
This limitation appears to be an arbitrary change and is not required by macOS standards. Returning to the previous view layout will better support user needs for multi-sensor monitoring.

**Proposed Solution:**

Revert sensor view functionality to the previous version (prior to v2.11.7) where multiple sensor statuses could be displayed simultaneously.

exelban, 2024-10-27:
have no idea what you are talking about

ukateki, 2024-10-27:
Before v2.11.7
[image]
After v2.11.7
[image]

exelban, 2024-10-27:
you are showing two different widgets. There is nothing to revert.

closed 2024-10-27 by exelban