exelban/stats #2199: Local IP reports "Unknown"

luckman212, 2024-10-29:
### Bug

In the network popup, my Local IP is reported as **Unknown** instead of the LAN address. Not sure when this started, but I noticed it today.

[image]

### Details

 - Device: M1 Mac Mini 2020
 - macOS: 15.1
 - Application version: 2.1.16

PuzzledUser, 2024-10-31:
Yup, me too.
Version 2.11.16, Mac OS 15.0.1, Mac mini M2

closed 2024-11-03 by exelban

luckman212, 2024-11-03:
Did I mis-file this? Not sure why it's closed

PuzzledUser, 2024-11-03:
Aha, good, I'm pleased it's come back. I had other suggestions filed like this and didn't like to nag about how you handle your project.

exelban, 2024-11-04:
@PuzzledUser you are free to do a fork and to do whatever you want there.

exelban, 2024-11-04:
@luckman212 local IP is from macOS API. Cannot do anything with that if it's not visible.

PuzzledUser, 2024-11-04:
`ifconfig | awk '/inet /&&!/127.0.0.1/{print $2;exit}'` or `ipconfig getifaddr en0` (or en1, en2 etc).
I hear you on the forking idea @exelban.

luckman212, 2024-11-04:
@exelban I think there's a bug somewhere. I haven't had enough time to figure it out, but switching the reader from "process based" to "interface based" is enough to get the LocalIP to update correctly. So the info is definitely there to be collected.

PuzzledUser, 2024-11-04:
That worked for me too @luckman212. Also, I swapped back to Process based and it continued to work (even after pressing reload ↻).

exelban, 2024-11-04:
@luckman212 hmm, yeap. If you are using the process based reader there is no network interface selected. Will check out this behavior. Thx for the info