exelban/stats #2200: Widget disappears on non-focused displays

josehu07, 2024-10-30:
**Describe the bug**

I have multiple displays, and am putting 5 Stats widgets on the menu bar. Stats works great and keeps refreshing on the menu bars on all displays, except that the right-most widget, `net`, disappears non-focused displays.

Screenshot of the display in focus:
[image]

Screenshot of the display if I focus on another monitor:
[image]

Not a main display vs. secondary displays issue. The widget always appears on the display of current focus and disappears on displays out of focus.

**Details:**
 - Device: Macbook M3 Pro 2023
 - macOS: 15.1
 - Application version: 2.11.16
 - Settings: `Combined modules` is on and `Combined details` is off

exelban, 2024-11-03:
Hi. What if you disable network module? Does the SSD will disappear?

josehu07, 2024-11-04:
Hi, thank you for the response! I tried several more combinations of modules and the results were interesting.

---
4, CPU + GPU + RAM + NET or SSD, all good:

[image]

[image]

---
5, CPU + GPU + RAM + (NET + SSD) or (SSD + NET), the last one disappears:

[image]

[image]

---
4, CPU + GPU + NET or SSD + RAM, the last one RAM disappears:

[image]

[image]

---
4, CPU + RAM + NET or SSD + GPU, the last one disappears:

[image]

[image]

---
5, CPU + RAM + SSD + NET + GPU, the last two both disappear:

[image]

[image]

---
5, CPU + SSD + NET + GPU + RAM, then both GPU and RAM disappear and there will be a gap:

[image]

[image]

---
3, CPU + NET + GPU, the last one disappears:

[image]

[image]

---
3, RAM + CPU + GPU, all good:

[image]

[image]

---
Overall, there seems to be a pattern: the one(s) that disappear are always those to the right of either NET or SSD, and sometimes they might overwrite modules in the middle.

I was using the line chart style with `label` on for both NET and SSD.

exelban, 2024-11-06:
thanks for such a wide debug)
Im pretty sure that has some relation with width. Will make a debug build to collect more info and will share it here.

josehu07, 2025-03-14:
I've got some free time today and tried debugging this myself. I think I've made some progress:

- After some `git bisect`, I found the offsetting behavior was introduced in this commit f2d9a49
- I'm currently looking into that commit (that introduced an updated version of `NetworkChartView`, which seems to be the source of the problem) -- will make a PR when I complete the fix
- I've also discovered a minor calculation bug in `CombinedView` that would draw separator lines (if turned on) incorrectly -- will have this fixed in the same PR

josehu07, 2025-03-14:
> After some `git bisect`, I found the offsetting behavior was introduced in this commit [f2d9a49](https://github.com/exelban/stats/commit/f2d9a491faded6243aef50162c3167246c04f2b4)

I've pinpointed the problem. A `context.restoreGState()` has been missed at https://github.com/exelban/stats/blob/master/Kit/Widgets/NetworkChart.swift#L207

Will make a PR soon.

PR #2444 (2025-03-16): fix: add missing NetworkChart restoreGState & correct CombinedView separators
Two very simple changes that fix the annoying issue of missing components on non-focused displays.

- Adding the missing `context.restoreGState()` in `NetworkChart` (fixes #2200)
- Correct separator drawing in `CombinedView` (fixes a wrong end condition on variable `i`)
closed 2025-03-16 by exelban