exelban/stats #2201: Many logs generated by Stats

PuzzledUser, 2024-10-31:
I note https://github.com/exelban/stats/issues/1728 was closed without comment. 

I just noticed this due to performance issues I was having. I have 7977 log entires from stats in 18 seconds. 2219 of these are Cache Delete messages.

- What's going on? 
- Is stats regularly clearing multiple caches?

Version 2.11.16, Mac OS 15.0.1, Mac mini M2

Short log excerpt below:
```default	09:03:08.816200+0000	Stats	323 CDRecentVolumeInfo _recentInfoAtUrgency, service: <private>, amount: 1019015168 <private>
default	09:03:08.816270+0000	Stats	CacheDeleteVolume effective_size: <private> container size: <private>, effective size: <private>
default	09:03:08.816287+0000	Stats	323 CDRecentVolumeInfo _recentInfoAtUrgency, service: <private>, amount: 65536 <private>
default	09:03:08.816346+0000	Stats	366 CDRecentVolumeInfo _recentInfoAtUrgency, volume: /System/Volumes/Data, urgency: 3, result:
default	09:03:08.816360+0000	Stats	com.apple.installd.CacheDelete : 0
default	09:03:08.816373+0000	Stats	com.apple.nsurlsessiond-cachedelete : 0
default	09:03:08.816385+0000	Stats	com.apple.quicklook.ThumbnailsAgent.CacheDelete : 1988902
default	09:03:08.816395+0000	Stats	com.apple.cloudd.cache-delete : 0
default	09:03:08.816406+0000	Stats	com.apple.triald.cache-delete : 0
default	09:03:08.816440+0000	Stats	com.apple.VIDEO : 0
default	09:03:08.816452+0000	Stats	com.apple.fspurgeable_data : 1019015168
default	09:03:08.816488+0000	Stats	com.apple.coresymbolicationd.cache-delete : 65536
default	09:03:08.816519+0000	Stats	com.apple.TMCacheDelete : 958943232
default	09:03:08.816535+0000	Stats	com.apple.mobileassetd.cache-delete : 38000000
default	09:03:08.816576+0000	Stats	com.apple.ReportMemoryException.CacheDelete : 0
default	09:03:08.816596+0000	Stats	com.apple.powerlog.CacheDelete : 0
default	09:03:08.816611+0000	Stats	com.apple.metadata.mds.cachedelete : 0
default	09:03:08.816649+0000	Stats	com.apple.cloudd.system.cache-delete : 0
default	09:03:08.816664+0000	Stats	com.apple.ibooks.CacheDeleteExtension : 0
default	09:03:08.816700+0000	Stats	CACHE_DELETE_SHARED_PURGEABLE : 32451504482
default	09:03:08.816715+0000	Stats	com.apple.geod.cachedelete : 0
default	09:03:08.816750+0000	Stats	com.apple.commerce.cachedelete : 0
default	09:03:08.816776+0000	Stats	com.apple.idleassetsd.CacheDelete : 28194918572
default	09:03:08.816802+0000	Stats	com.apple.AssetCache.builtin.CacheDelete : 86016
default	09:03:08.816829+0000	Stats	com.apple.bird.cache-delete : 0
default	09:03:08.816858+0000	Stats	com.apple.Music.MusicCacheExtension : 0
default	09:03:08.816884+0000	Stats	CACHE_DELETE_VOLUME : /System/Volumes/Data
default	09:03:08.816910+0000	Stats	com.apple.cache_delete_orphan_dir_handler : 0
default	09:03:08.816937+0000	Stats	com.apple.aned.CacheDelete : 0
default	09:03:08.816963+0000	Stats	com.apple.MailCacheDelete : 0
default	09:03:08.816988+0000	Stats	com.apple.TV.TVCacheExtension : 0
default	09:03:08.817003+0000	Stats	com.apple.appstoreagent.CacheDelete : 0
default	09:03:08.817039+0000	Stats	com.apple.aneuserd.CacheDelete : 0
default	09:03:08.817065+0000	Stats	com.apple.replayd-cache-delete : 64
default	09:03:08.817090+0000	Stats	com.apple.MOVIE : 0
default	09:03:08.817116+0000	Stats	com.apple.wallpaper.CacheDelete : 0
default	09:03:08.817146+0000	Stats	com.apple.MUSIC : 0
default	09:03:08.817178+0000	Stats	com.apple.logd.cachedelete : 1103686096
```
and
```
default	09:03:08.817956+0000	Stats	com.apple.appstoreagent.CacheDelete : {
    "CACHE_DELETE_PROCNAME" = appstoreagent;
    "CACHE_DELETE_SIGNING_ID" = "com.apple.appstoreagent";
}
default	09:03:08.817971+0000	Stats	com.apple.aneuserd.CacheDelete : {
    "CACHE_DELETE_PROCNAME" = "";
    "CACHE_DELETE_SIGNING_ID" = "com.apple.aneuserd";
}
default	09:03:08.818006+0000	Stats	com.apple.replayd-cache-delete : {
    "CACHE_DELETE_PROCNAME" = replayd;
    "CACHE_DELETE_SIGNING_ID" = "com.apple.replayd";
}
default	09:03:08.818022+0000	Stats	com.apple.MOVIE : {
    "CACHE_DELETE_PROCNAME" = "";
    "CACHE_DELETE_SIGNING_ID" = "com.apple.deleted_helper";
}
default	09:03:08.818053+0000	Stats	com.apple.wallpaper.CacheDelete : {
    "CACHE_DELETE_PROCNAME" = WallpaperAgent;
    "CACHE_DELETE_SIGNING_ID" = "com.apple.wallpaper.agent";
}
default	09:03:08.818083+0000	Stats	com.apple.MUSIC : {
    "CACHE_DELETE_PROCNAME" = "";
    "CACHE_DELETE_SIGNING_ID" = "com.apple.deleted_helper";
}
default	09:03:08.818099+0000	Stats	com.apple.logd.cachedelete : {
    "CACHE_DELETE_PROCNAME" = "";
    "CACHE_DELETE_SIGNING_ID" = "com.apple.logd";
}
default	09:03:08.818135+0000	Stats	com.apple.imagent.cache-delete : {
    "CACHE_DELETE_PROCNAME" = "";
    "CACHE_DELETE_SIGNING_ID" = "com.apple.deleted_helper";
}
default	09:03:08.818151+0000	Stats	com.apple.Safari.CacheDeleteExtension : {
    "CACHE_DELETE_PROCNAME" = CacheDeleteExtension;
    "CACHE_DELETE_SIGNING_ID" = "com.apple.Safari.CacheDeleteExtension";
}
```

PuzzledUser, 2024-10-31:
Perplexity confirms that this seems odd and potential overreach: https://www.perplexity.ai/search/what-doe-this-mean-in-mac-os-l-kCz1wQKUSka85RrRqPFz4A

**Please give this some thought Serhiy before closing.**

exelban, 2024-10-31:
There is nothing related to the Stats itself. It's a MacOS log that is used to obtain disk space information. If you don't want to disable these logs just disable the Disk module. There is nothing I can do about that. But if you have any idea of what could be done here PR is welcomed.

PuzzledUser, 2024-10-31:
I understand and appreciate your explanation (and reassurance). I'll post back if I think of anything.

closed 2024-11-03 by exelban