exelban/stats #2204: Change battery icon when charging

gelpenaddict, 2024-11-06:
Currently, the battery icon stays the same when plugged in and charging.

Can it be updated to indicate that it's charging?

Examples:
- https://images.app.goo.gl/iXmViZPFvbDbYCVW9
- https://images.app.goo.gl/u1b5mkf4QW7D6LDL6

It'd be useful to see immediately if an outlet isn't working, especially when traveling.

exelban, 2024-11-06:
There are 2 different icons already: power plug only when there is no charging and lightning when charging is active.

closed 2024-11-06 by exelban

gelpenaddict, 2024-11-06:
[image]

I'm not seeing it, is something not working then?

exelban, 2024-11-06:
if you want to see battery icons you need to enable battery widget

pyro-gx, 2024-11-14:
Switched to https://github.com/gao-sun/eul for this feature.