exelban/stats #2208: Continuous notification Swap file every second

warmup72, 2024-11-09:
If you configure Stats to send a message as soon as a swap file is created, you will be spammed with messages. As soon as the event occurs, you will receive a message every second.
As a swap file can hardly be reduced in size, for me this means either stopping Stats or continuing to be spammed.

An ignore nn-minute button would be helpful

exelban, 2024-11-09:
or you can just disable the swap size notification. New notification will not be shown if: the previous was dismissed or the previous state was changed to the normal state.

warmup72, 2024-11-09:
Thanks for the tip, but that doesn't help with the problem.
See it as a feature request. 
I would like to be notified when my swap reaches a certain size, but not every second, but at longer intervals

exelban, 2024-11-09:
And it must be like that. The only way that you have repeating notification is that your swap goes lower that threshold, and then notification disappear. And appear when it goes more than threshold.

exelban, 2024-11-09:
Is the notification blinking? I mean appear and disappear, or you have a tons of notification stacked?

warmup72, 2024-11-09:
The message appears and disappears, but the size of the swap file doesn't changes in the message.
The messages are then stacked.

exelban, 2024-11-10:
If appear and disappear it means that value goes below threshold. I will make a debug build for you to debug that.

warmup72, 2024-11-10:
Thank you for that, because it can't really be. I have set a notification at 50MB. As far as I know, a swap file simply grows on a Mac. I have not yet noticed a significant reduction.
Unfortunately, I decided in favour of ‘only’ 18GB for my M§, so I have to deal with the swap :-)

exelban, 2024-11-10:
[Stats.dmg.zip](https://github.com/user-attachments/files/17692531/Stats.dmg.zip)

This build writes some logs to the log.txt that could be found in the Documents folder.

warmup72, 2024-11-10:
Guten abend, ich habe das gleich mal versucht. Am Anfang sah alles gut aus, gleich nachdem ich 1,7GB Swap hatte ging es los. Im Sekundentakt Meldungen. Gerade eben bekomme ich im Sekundentakt 1,5GB-Nachrichten.
Das witzige, wenn ich versuche die Meldungen mit Omni Recorder aufzunehmen verschwinden sie, bis ich die Aufnahme beende :-)
[log.txt](https://github.com/user-attachments/files/17692680/log.txt)

exelban, 2024-11-10:
sorry, but could you use English please?

exelban, 2024-11-11:
please record a video of notification behavior. Cause I cannot reproduce that(
and cannot understand what causing that.

warmup72, 2024-11-11:
Unfortunately, I can't make a screen recording of the problem because the messages stop as soon as I start the recording.
Here is a video of the problem recorded with my mobile phone

https://github.com/user-attachments/assets/13282507-b4af-4ab5-9c65-7a4a42c608a6

exelban, 2024-11-11:
hmm, thx. Will try to reproduce that. It's definitely looks like notification disappear and appear again...

warmup72, 2024-11-11:
I use brave :-) to fill up the memory. Just open 10 windows in which 10 videos are played.  After the swap message appears for the first time, simply start closing the tabs until the messages start to run.

exelban, 2024-11-11:
[Stats.dmg.zip](https://github.com/user-attachments/files/17706556/Stats.dmg.zip)

try please this build and share the logs

warmup72, 2024-11-11:
Many thanks for the quick reply. 
I will be away from my Mac until Thursday. I'll get back to you as soon as I'm back on

warmup72, 2024-11-15:
Good evening, unfortunately I only got round to testing it today. Here is the log - at 17:35 the messages started flooding in again
[log.txt](https://github.com/user-attachments/files/17778415/log.txt)

exelban, 2024-11-16:
Hi, thanks. Hmm, very interesting, looks like some bug with notification id.

exelban, 2024-11-16:
[Stats.dmg.zip](https://github.com/user-attachments/files/17784291/Stats.dmg.zip)

warmup72, 2024-11-16:
Thank you very much,
with this version, Stats has become quiet.
No matter how I developed the size of the swap file I only got a message at 212MB.
In the meantime, the file has grown to 2.5GB and also become smaller again, but unfortunately without any further message

exelban, 2024-11-17:
thx for feedback and for helping in debugging

warmup72, 2024-11-17:
Thanks for your help - but I think more news, especially as the swap file grows would be desirable.