exelban/stats #2212: Stats v2.11.17 crashes immediately after starting

fad1, 2024-11-11:
Widgets appear for like 1/10 of a second then disappear. I tried starting from terminal to get an error message, see below:

**Details:**
 - Device: MBA2020 M1
 - macOS: Sonoma 14.7.1
 - Application version: 2.11.17

❯ /Applications/Stats.app/Contents/MacOS/Stats
2024-11-11 15:10:16 reader.swift:85 DBG [<CPU.LoadReader: 0x137f138c0>] Successfully initialize reader
2024-11-11 15:10:16 reader.swift:159 DBG [<CPU.ProcessReader: 0x60000006da70>] Set update interval: 5 sec
2024-11-11 15:10:16 reader.swift:85 DBG [<CPU.ProcessReader: 0x60000006da70>] Successfully initialize reader
2024-11-11 15:10:16 reader.swift:159 DBG [<CPU.AverageReader: 0x60000006ddd0>] Set update interval: 60 sec
2024-11-11 15:10:16 reader.swift:85 DBG [<CPU.AverageReader: 0x60000006ddd0>] Successfully initialize reader
2024-11-11 15:10:16 reader.swift:85 DBG [<CPU.TemperatureReader: 0x60000006de60>] Successfully initialize reader
2024-11-11 15:10:16 reader.swift:85 DBG [<GPU.InfoReader: 0x6000000643f0>] Successfully initialize reader
2024-11-11 15:10:16.565 Stats[21887:230699] NSPanel requires NSWindowStyleMaskUtilityWindow for a HUD window
2024-11-11 15:10:16 reader.swift:85 DBG [<RAM.UsageReader: 0x600000d53930>] Successfully initialize reader
2024-11-11 15:10:16 reader.swift:159 DBG [<RAM.ProcessReader: 0x60000005a7f0>] Set update interval: 1 sec
2024-11-11 15:10:16 reader.swift:85 DBG [<RAM.ProcessReader: 0x60000005a7f0>] Successfully initialize reader
2024-11-11 15:10:16 reader.swift:85 DBG [<Disk.CapacityReader: 0x600001c02c00>] Successfully initialize reader
2024-11-11 15:10:16 reader.swift:159 DBG [<Disk.ActivityReader: 0x600001c02e00>] Set update interval: 1 sec
2024-11-11 15:10:16 reader.swift:85 DBG [<Disk.ActivityReader: 0x600001c02e00>] Successfully initialize reader
2024-11-11 15:10:16 reader.swift:159 DBG [<Disk.ProcessReader: 0x6000000650e0>] Set update interval: 1 sec
2024-11-11 15:10:16 reader.swift:85 DBG [<Disk.ProcessReader: 0x6000000650e0>] Successfully initialize reader
2024-11-11 15:10:16 reader.swift:85 DBG [<Sensors.SensorsReader: 0x60000036e300>] Successfully initialize reader
2024-11-11 15:10:17.093 Stats[21887:230699] NSPanel requires NSWindowStyleMaskUtilityWindow for a HUD window
2024-11-11 15:10:17 reader.swift:85 DBG [<Net.UsageReader: 0x13913e970>] Successfully initialize reader
2024-11-11 15:10:17 reader.swift:85 DBG [<Net.ProcessReader: 0x600000048000>] Successfully initialize reader
2024-11-11 15:10:17 reader.swift:85 DBG [<Net.ConnectivityReader: 0x130375fc0>] Successfully initialize reader
2024-11-11 15:10:17 reader.swift:85 DBG [<Battery.UsageReader: 0x1304ae9d0>] Successfully initialize reader
2024-11-11 15:10:17 reader.swift:85 DBG [<Battery.ProcessReader: 0x600001ca9880>] Successfully initialize reader
2024-11-11 15:10:17 reader.swift:85 DBG [<Bluetooth.DevicesReader: 0x60000035d540>] Successfully initialize reader
2024-11-11 15:10:17 reader.swift:85 DBG [<Clock.ClockReader: 0x600001c9a600>] Successfully initialize reader
Error Domain=NSCocoaErrorDomain Code=4099 "The connection to service named eu.exelban.Stats.SMC.Helper was invalidated: failed at lookup with error 3 - No such process." UserInfo={NSDebugDescription=The connection to service named eu.exelban.Stats.SMC.Helper was invalidated: failed at lookup with error 3 - No such process.}
2024-11-11 15:10:17 reader.swift:128 DBG [<Net.UsageReader: 0x13913e970>] Set up update interval: 1 sec
2024-11-11 15:10:17 reader.swift:128 DBG [<Net.ConnectivityReader: 0x130375fc0>] Set up update interval: 1 sec
2024-11-11 15:10:17 widget.swift:296 DBG [Network] widget speed enabled
2024-11-11 15:10:17 reader.swift:128 DBG [<Disk.CapacityReader: 0x600001c02c00>] Set up update interval: 1 sec
2024-11-11 15:10:17 reader.swift:128 DBG [<Disk.ActivityReader: 0x600001c02e00>] Set up update interval: 1 sec
2024-11-11 15:10:17 widget.swift:296 DBG [Disk] widget mini enabled
2024-11-11 15:10:17 reader.swift:128 DBG [<RAM.UsageReader: 0x600000d53930>] Set up update interval: 5 sec
2024-11-11 15:10:17 widget.swift:296 DBG [RAM] widget pie_chart enabled
2024-11-11 15:10:17 widget.swift:296 DBG [RAM] widget text enabled
2024-11-11 15:10:17 reader.swift:128 DBG [<CPU.LoadReader: 0x137f138c0>] Set up update interval: 5 sec
2024-11-11 15:10:17 widget.swift:296 DBG [CPU] widget bar_chart enabled
2024-11-11 15:10:17 helpers.swift:102 DBG Application update interval is 'Silent'
2024-11-11 15:10:17 helpers.swift:143 DBG error updater.check(): last check was 2 minutes ago, stopping...
2024-11-11 15:10:17 AppDelegate.swift:74 INF Stats started in 0.0054 seconds
dyld[21887]: missing symbol called
[1]    21887 abort      /Applications/Stats.app/Contents/MacOS/Stats

The problem disappears when I install .16, using pearcleaner to delete all files related to Stats and reinstall does not prevent the issue.

exelban, 2024-11-11:
the log is totally unusable

miss4chewN8, 2024-11-11:
I have had the same issue while using the 2.11.17 upgrade and enabling the new RAM text widget. Upon reinstalling and installing using brew, the issue persisted as I suspect the settings stayed the same. I had to do a clean install using pearcleaner whilst still using 2.11.17 and not selecting the RAM text widget and things appear to be working now.

fad1, 2024-11-11:
I too enabled the RAM text widget, I'll retry cleaning everything.

exelban, 2024-11-11:
hmm. Could someone share crash report?

fad1, 2024-11-12:
Sure

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [21887]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.11.17 (609)
Code Type:             ARM-64 (Native)
Parent Process:        zsh [1042]
Responsible:           iTerm2 [932]
User ID:               501

Date/Time:             2024-11-11 15:10:17.3936 +0100
OS Version:            macOS 14.7.1 (23H222)
Report Version:        12
Anonymous UUID:        A0559AB3-A82A-F827-7DD5-68D317CC59C0

Sleep/Wake UUID:       ECC8161A-A1BF-4B62-9EDC-95BBFDB66BA3

Time Awake Since Boot: 9500 seconds
Time Since Wake:       1172 seconds

System Integrity Protection: enabled

Crashed Thread:        8  Dispatch queue: com.apple.root.background-qos

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 9 
missing symbol called

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_platform.dylib      	       0x19ecadf5c _platform_memset + 140
1   CoreGraphics                  	       0x1a4927004 aa_cache_render + 544
2   CoreGraphics                  	       0x1a4922fac CGGlyphBitmapCreateWithPathAndDilation_8 + 884
3   CoreGraphics                  	       0x1a4920458 CGFontCreateGlyphBitmapWithDilation + 628
4   CoreGraphics                  	       0x1a4920150 CGGlyphBuilder::create_missing_bitmaps(CGGlyphIdentifier const*, unsigned long, CGGlyphBitmap const**) + 128
5   CoreGraphics                  	       0x1a491fccc CGGlyphBuilder::lock_glyph_bitmaps(CGGlyphIdentifier const*, unsigned long, CGGlyphBitmap const**) + 160
6   CoreGraphics                  	       0x1a491fadc render_glyphs + 236
7   CoreGraphics                  	       0x1a491f2e0 draw_glyph_bitmaps + 1072
8   CoreGraphics                  	       0x1a491ee04 ripc_DrawGlyphs + 1268
9   CoreGraphics                  	       0x1a491e780 CGContextDelegateDrawGlyphs + 368
10  CoreGraphics                  	       0x1a497e2bc CG::DisplayListExecutor::drawGlyphs(CG::DisplayListEntryGlyphs const*) + 464
11  CoreGraphics                  	       0x1a4d5b3a8 CG::DisplayList::executeEntries(std::__1::__wrap_iter<std::__1::shared_ptr<CG::DisplayListEntry const>*>, std::__1::__wrap_iter<std::__1::shared_ptr<CG::DisplayListEntry const>*>, CGContextDelegate*, CGRenderingState*, CGGStack*, CGRect const*, __CFDictionary const*, bool) + 188
12  CoreGraphics                  	       0x1a497b51c CG::DisplayList::execute(CGContextDelegate*, CGRenderingState*, CGGStack*, CGRect const*, __CFDictionary const*) + 296
13  CoreGraphics                  	       0x1a497b2fc CGDisplayListDrawInContextDelegate + 528
14  QuartzCore                    	       0x1a6f71c48 CABackingStoreUpdate_ + 500
15  QuartzCore                    	       0x1a6fc9490 invocation function for block in CA::Layer::display_() + 64
16  QuartzCore                    	       0x1a6f70cc0 -[CALayer _display] + 1636
17  AppKit                        	       0x1a29728ac 0x1a2581000 + 4135084
18  AppKit                        	       0x1a2972320 0x1a2581000 + 4133664
19  AppKit                        	       0x1a297293c 0x1a2581000 + 4135228
20  QuartzCore                    	       0x1a6f6fe40 CA::Layer::display_if_needed(CA::Transaction*) + 744
21  QuartzCore                    	       0x1a70f4cd8 CA::Context::commit_transaction(CA::Transaction*, double, double*) + 512
22  QuartzCore                    	       0x1a6f52864 CA::Transaction::commit() + 648
23  AppKit                        	       0x1a26e3ac8 __62+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayLink]_block_invoke + 272
24  AppKit                        	       0x1a30a44e4 ___NSRunLoopObserverCreateWithHandler_block_invoke + 64
25  CoreFoundation                	       0x19ed5c85c __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
26  CoreFoundation                	       0x19ed5c748 __CFRunLoopDoObservers + 536
27  CoreFoundation                	       0x19ed5bd74 __CFRunLoopRun + 776
28  CoreFoundation                	       0x19ed5b414 CFRunLoopRunSpecific + 608
29  HIToolbox                     	       0x1a950619c RunCurrentEventLoopInMode + 292
30  HIToolbox                     	       0x1a9505e2c ReceiveNextEventCommon + 220
31  HIToolbox                     	       0x1a9505d30 _BlockUntilNextEventMatchingListInModeWithFilter + 76
32  AppKit                        	       0x1a25bacc8 _DPSNextEvent + 660
33  AppKit                        	       0x1a2db14d0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
[... 782 lines cut ...]
  "rollouts" : [
    {
      "rolloutId" : "64628732bf2f5257dedc8988",
      "factorPackIds" : {

      },
      "deploymentId" : 240000001
    },
    {
      "rolloutId" : "645197bf528fbf3c3af54105",
      "factorPackIds" : {
        "SIRI_VALUE_INFERENCE_PERVASIVE_ENTITY_RESOLUTION" : "663e65b4a1526e1ca0e288a1"
      },
      "deploymentId" : 240000002
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "45f4e2a5-551b-4bc2-a2dc-19c244dda8f8",
      "experimentId" : "6643969b3099cf28e049862f",
      "deploymentId" : 400000007
    }
  ]
}
}

exelban, 2024-11-12:
thx. Could you please try this build:

[Stats.dmg.zip](https://github.com/user-attachments/files/17714806/Stats.dmg.zip)

fad1, 2024-11-12:
Sure, crashes too:
[c4.txt](https://github.com/user-attachments/files/17714839/c4.txt)
[c5.txt](https://github.com/user-attachments/files/17714841/c5.txt)

fad1, 2024-11-12:
1st crash was after trying immediately, 2nd and 3rd after deleting everything with pearcleaner.
[Stats-2024-11-12-115240.txt](https://github.com/user-attachments/files/17715395/Stats-2024-11-12-115240.txt)
[Stats-2024-11-12-115422.txt](https://github.com/user-attachments/files/17715394/Stats-2024-11-12-115422.txt)
[Stats-2024-11-12-115455.txt](https://github.com/user-attachments/files/17715393/Stats-2024-11-12-115455.txt)

exelban, 2024-11-12:
thx, and what about this one?
PS: one crash report is enough

[Stats.dmg.zip](https://github.com/user-attachments/files/17715499/Stats.dmg.zip)

fad1, 2024-11-12:
crashes [Stats-2024-11-12-120610.txt](https://github.com/user-attachments/files/17715538/Stats-2024-11-12-120610.txt)

exelban, 2024-11-12:
cannot understand, I disable almost everything. Try this build:
[Stats.dmg.zip](https://github.com/user-attachments/files/17715919/Stats.dmg.zip)

fad1, 2024-11-12:
This one did not crash, but check how RAM text is shown:

[image]

[image]

exelban, 2024-11-12:
yeap, it's because I have disable everything related to that widget. Now I will enable step by step features and will see what causing the problem.

[Stats.dmg.zip](https://github.com/user-attachments/files/17716147/Stats.dmg.zip)

fad1, 2024-11-12:
Works
[image]

exelban, 2024-11-12:
thx, thank you for helping debug the problem)

[Stats.dmg.zip](https://github.com/user-attachments/files/17716248/Stats.dmg.zip)

fad1, 2024-11-12:
No problem. Works

[image]

[image]

exelban, 2024-11-12:
[Stats.dmg.zip](https://github.com/user-attachments/files/17716432/Stats.dmg.zip)

fad1, 2024-11-12:
[image]
[image]

fad1, 2024-11-12:
[Stats-2024-11-12-135515.txt](https://github.com/user-attachments/files/17716886/Stats-2024-11-12-135515.txt)

fad1, 2024-11-12:
[image]

exelban, 2024-11-12:
[Stats.dmg.zip](https://github.com/user-attachments/files/17717090/Stats.dmg.zip)

fad1, 2024-11-12:
[image]
[image]

exelban, 2024-11-12:
[Stats.dmg.zip](https://github.com/user-attachments/files/17717201/Stats.dmg.zip)

exelban, 2024-11-12:
[Stats.dmg.zip](https://github.com/user-attachments/files/17717352/Stats.dmg.zip)

fad1, 2024-11-12:
[image]

624

exelban, 2024-11-12:
looks like pressure parser makes problem(
takes too much time to find that(

[Stats.dmg.zip](https://github.com/user-attachments/files/17717444/Stats.dmg.zip)

fad1, 2024-11-12:
[image]
625

exelban, 2024-11-12:
[Stats.dmg.zip](https://github.com/user-attachments/files/17717694/Stats.dmg.zip)

fad1, 2024-11-12:
[Stats-2024-11-12-153236.txt](https://github.com/user-attachments/files/17718063/Stats-2024-11-12-153236.txt)

exelban, 2024-11-12:
[Stats.dmg.zip](https://github.com/user-attachments/files/17718311/Stats.dmg.zip)

this build will write some logs to the log.txt that could be found in the Documents folder. Could you please share it after running that build?

exelban, 2024-11-12:
so this build does not even run?

fad1, 2024-11-12:
It does not

exelban, 2024-11-12:
[Stats.dmg.zip](https://github.com/user-attachments/files/17719557/Stats.dmg.zip)

exelban, 2024-11-12:
or this one:

[Stats.dmg.zip](https://github.com/user-attachments/files/17720324/Stats.dmg.zip)

fad1, 2024-11-12:
> [Stats.dmg.zip](https://github.com/user-attachments/files/17719557/Stats.dmg.zip)

[image]
629

fad1, 2024-11-12:
> or this one:
> 
> [Stats.dmg.zip](https://github.com/user-attachments/files/17720324/Stats.dmg.zip)

[Stats-2024-11-12-190916.txt](https://github.com/user-attachments/files/17721011/Stats-2024-11-12-190916.txt)

exelban, 2024-11-12:
[Stats.dmg.zip](https://github.com/user-attachments/files/17722087/Stats.dmg.zip)

fad1, 2024-11-12:
[image]
631

exelban, 2024-11-12:
wow, nice

fad1, 2024-11-12:
All my widgets seem to be working. I'm using $swap.used as it is what I understand best to represent memory pressure.

I'll make sure to check the hotfix commit to see what was the cause of the problem ;)

fad1, 2024-11-14:
I just noticed this file
[log.txt](https://github.com/user-attachments/files/17745916/log.txt)

exelban, 2024-11-14:
this API was used for a years already

closed 2024-11-14 by exelban