exelban/stats #2215: Add vertical bar between modules

UsamaKarim, 2024-11-13:
[image]

The menu bar on the right side comes from the Tencent Lemon app. It features vertical bars that make statistics easy to view. Can you please add similar functionality to the stats?

I'd like to retain the use of stats, as it offers more versatility and is easier to configure compared to other options. Additionally, stats provide detailed information about each module, making them highly customizable. However, this level of customization can sometimes make the statistics difficult to read. To address this issue, I've added a reasonable amount of space between modules, which has improved readability somewhat. Nevertheless, I still find it challenging to view the stats clearly. I believe incorporating vertical bars would greatly enhance the readability of these sections.

exelban, 2024-11-13:
hi, thanks for the feedback. I suppose you are talking about combined modules view?

UsamaKarim, 2024-11-13:
> hi, thanks for the feedback. I suppose you are talking about combined modules view?

Yes, you got it right.

exelban, 2024-11-13:
yeap, I like that and I think it could be added

closed 2024-12-15 by exelban

UsamaKarim, 2024-12-16:
Thanks for the implementation. I noticed a minor issue with the bar color.

Currently, the implementation is based on the appearance (dark mode and white mode), but the bar colors remain unchanged when the appearance mode is switched (from dark mode to light mode and vice versa). I have auto dark mode enabled and observed this issue.
The black bar colors appear unusual, and to resolve this issue, simply hard-coding white bars would suffice.
If you wish to expand the implementation, a color picker could also be added.

UsamaKarim, 2024-12-16:
I found no bars between few modules
[image]