exelban/stats #2222: Add visibility of 'bad' swap-out (and swap-in) activity

PuzzledUser, 2024-11-15:
SSDs age when repeatedly written to, so whilst static swap usage is kinda ok, active _swapping out_ is bad and worth avoiding. _Swapping-in_ is not really an issue as it doesn't age SSDs.

1. Could you add $swap.out (and $swap.in) to the text widget value in RAM, Module?
2. Better yet, could you add an option to select what the RAM module graphs, (or add a new Swap module) to show swap.out (or swap.in, or swap.used)?

**vm_stat** provides these figures. Maybe there is a better way of notifying when lots of active swapping (out) occurs? 

Thanks for considering.

exelban, 2024-11-18:
I think the swapins and swapouts could be added to the text parser in RAM widget.

closed 2024-12-01 by exelban

PuzzledUser, 2024-12-02:
Thank you. Is there as way of graphing or getting notification based on swapouts?

ChillarAnand, 2025-02-10:
Any plans to add widget for swapin/outs? The text value is consuming a lot of space in the menu bar.

[image]