exelban/stats #2225: External SSD temperature monitoring

ctrl-shift-vs, 2024-11-16:
**Feature request**

I would like to see my external SSD temperature as part of other sensor suite in menubar. The external SSD temperature can already be seen when in SSD menubar icon after clicking on the control icon (see image), but cannot be added to the Sensors icon suite (only one option for NAND, which must be internal storage). Please add this feature. Thanks!!

[image]

exelban, 2024-11-18:
It's not possible. This value is available only in Disk module. And it cannot be added to the Sensors module.

closed 2024-11-18 by exelban

beckxie, 2025-02-04:
Hi @ctrl-shift-vs , 

I also had the same need to display disk temperature.
Since the current sensor widget does not support it, I made a simple implementation by modifying the Disk widget to show temperature instead of usage.

[image]

However, my approach is an intrusive modification, so it’s not suitable for merging into the upstream repository. 
Perhaps @exelban could consider officially adding this feature to the Disk module in the future.

For reference, here are the details: https://github.com/beckxie/stats/pull/1