exelban/stats #2229: Wrong IP Location

SamadiPour, 2024-11-20:
**Describe the bug**
After noticing that my IP is listed as "US", I wondered if there might be an issue with the provider that maps IP addresses to locations. I checked the code and found out that you retrieve the country using the `/usr/sbin/system_profiler SPAirPortDataType` command. I did some research and asked ChatGPT, and although I couldn't find any documentation to support this, I believe the country in the response does not correspond to the IP address; instead, it relates to the broadcast country code of the Wi-Fi.
Here is an example discussion related to this parameter: https://community.tp-link.com/en/home/forum/topic/174670.

**Details:**
 - Device: Macbook Pro M1
 - macOS: 15.1.1
 - Application version: 2.11.16

Also related to #2183

exelban, 2024-11-20:
so, what do you propose?

SamadiPour, 2024-11-20:
I don't have a solution for your self-hosted API, but you can use a free API that doesn't require an API key.

For Example:
- https://freeipapi.com
- https://ip-api.com

Every service I know has a rate limit, but since you are sending the request from the user's device, it shouldn't matter.

exelban, 2024-11-24:
Since there is no better solution than the native one, it looks like nothing could be changed. Ping me if someone has a better solution than the existing one.

closed 2024-11-24 by exelban

SamadiPour, 2024-11-27:
@exelban What if we use https://www.cloudflare.com/cdn-cgi/trace instead? We can replace the current API in the code with this one, as it provides the IP address and location.

NikolaiT, 2024-12-03:
You could always use an accurate service such as https://ipapi.is/ with an extensive free plan