exelban/stats #2230: Incorrect conversion between bps, kbps, mbps

dwilliams-1, 2024-11-20:
**Describe the bug**
It looks like a factor of 1024 is being used to convert between bps, kbps, mbps, gbps. While appropriate for storage/memory, network throughput needs to be scaled as a factor of 1000 (1kbps = 1000bps, 1mbps = 1000000bps, etc). This would also be applicable for B, KB, MB, GB as well. 

**Details:**
 - Device: [e.g. Macbook Pro M1]
 - macOS: [e.g. 15.1]
 - Application version: [e.g. 2.11.18]

exelban, 2024-11-20:
wat?

dwilliams-1, 2024-11-20:
To add an example
If you start an iperf3 session using '-b 100m' to send 100Mbps, iperf3 will report 100Mbps.
-open activity monitor in and view network activity. You should see ~13.1MB/s (104.8Mbps - the byte counters on a system read with overhead included)

Looking at the stats application
-In application mode -> open stats (network in Bytes to match activity mon). You will see ~11.9MB/s (95.2Mbps). If we undo the use of 1024 and replace it with 1000 we will see expected values
95.2*1024*1024 / 1000000 = 99.8Mbps -- expected value

-In interface mode -> open stats (network in Bytes to match activity mon). You will see ~12.5MB/s (100Mbps). If we undo the use of 1024 and replace it with 1000 we will see expected values
100*1024*1024 / 1000000 =  104.9Mbps -- expected value

When scaling network speeds, you use multiples of 1000, not 1024. Network speeds are a metric conversion vs storage/memory being based on binary multiples. Currently the network information is calculating Mebi(bits/bytes) per second instead of Mega(bits/bytes) per second. Both are valid numbers, however Mega is what is reflected by speedtest applications, os network tabs, etc. I am slow to trace things out here, but I think it stems from the reuse of the the units or getReadableMemory helpers.

fiwswe, 2024-11-20:
To be pedantic this is the old discussion about metric vs binary units. KB vs KiB, MB vs MiB, etc. Originally computer people used the units wrong, i.e. to mean 1 KB = 1024 Bytes.

If the current numbers use the binary (1024) multiplier then the units should be KiB, MiB, GiB, …

See https://en.wikipedia.org/wiki/Kilobyte

I have no opinion on what values should be displayed. But it should be noted that practically all marketing material and end-user documentation, including the Finder on macOS, have been using the metric units (1 KB = 1000) for quite a while now. The only notable exception seems to be RAM size as no one would understand why a machine has e.g. 8.59 GB of RAM instead of 8 GB (actually 8 GiB).

exelban, 2024-11-21:
thx, I will take a look at this later today

exelban, 2024-12-17:
I don't even know that macOS uses metric and not binary system. I will change that in the Stats to fit the macOS calculations.

closed 2024-12-22 by exelban