exelban/stats #2246: Just released v2.11.19 doesn't start on Mac mini M4

DigiH, 2024-12-01:
**Describe the bug**
Version 2.11.19 doesn't start at all on my Mac mini M4. Retried 2.11.18 which starts up fine with my previously used settings.

**Details:**
 - Device: Mac mini M4
 - macOS: 15.1.1
 - Application version: 2.11.19

898803709, 2024-12-01:
The same thing happened to me

DigiH, 2024-12-01:
Just saw, already logged

https://github.com/exelban/stats/issues/2245

luckman212, 2024-12-01:
Yes it would be good if these issues were merged.

felk-dev, 2024-12-01:
NO!!! I wait it for looooooooooong time:(

exelban, 2024-12-01:
the icon was ok, just wrong icon name(
Please try this build:

[Stats.dmg.zip](https://github.com/user-attachments/files/17969844/Stats.dmg.zip)

felk-dev, 2024-12-01:
Wow so fast bug fixed. thanks!!

exelban, 2024-12-01:
@felk-dev ok, thx for feedback. I have updated the release with a new build. So the fixed build could be downloaded there.

DigiH, 2024-12-01:
> the icon was ok, just wrong icon name( Please try this build:
> 
> [Stats.dmg.zip](https://github.com/user-attachments/files/17969844/Stats.dmg.zip)

Verified as fixed with that amended build. Closing.

closed 2024-12-01 by DigiH