exelban/stats #2249: Sensors Are Doesnt Work

basercagan, 2024-12-01:
cpu and gpu temperature sensors are not showing

 - Device: [Macbook Pro 2024 M4]
 - macOS: [15.1.1]
 - Application version: [2.11.19]

therealFoxster, 2024-12-01:
Same here on M4 MBP. Might be something with the latest update

fran-6co, 2024-12-02:
Same on Mac mini M4

exelban, 2024-12-02:
right now there is no sensors list for m4. I'm interested in what sensors have you seen before. Today I will prepare a few builds that allow detect what m4 is most similar to.

fran-6co, 2024-12-02:
This was showed on previous build 2.11.18. Note I marked some duplicated names. Also missing some, i.e.: CPU efficiency core 2 & 3, CPU performance core 4, 5 &  6.

[image]

PatDal81, 2024-12-03:
Weird, I do have temps on 2.11.19 (M4 Max, 16 cores). But I don't have Average CPU/GPU.

[image]

reedog117, 2024-12-03:
Probably related, but my temperature sensor is stuckat 212 F / 100 C for my CPU. This is for a base M4 inside a Macbook Pro 14".
[image]

exelban, 2024-12-03:
yes, basically it looks like Apple has changed the sensor keys again. If there is someone who wants to help I would recommend taking a look [here](https://github.com/exelban/stats/issues/1772) how it could be done.

Since I do not have an m4 mac I cannot help with that process.

PatDal81, 2024-12-03:
Willing to help, but I don't know where to start exactly.  Seem like the script you're refering to is for M3 only.  Any other hint I should be looking at?

exelban, 2024-12-03:
it was a generic script for any CPU

markop90, 2024-12-03:
I don't know if I ran the script correctly, but this is the result. In the middle of the script, I ran a benchmark tool to stress the cpu, then I stop it.

[m4.txt](https://github.com/user-attachments/files/17999217/m4.txt)

nilthacker, 2024-12-04:
I can run it for the 12 core M4 pro if someone can give me step by step instructions. The linked thread is just too long to follow.

EriksRemess, 2024-12-05:
If it helps, IOHIDService reports these temperature sensors with PrimaryUsagePage 0xff00, PrimaryUsage 5:

NAND CH0 temp
PMU tcal
PMU tdev{1..8}
PMU tdie{1..14}
PMU2 tcal
PMU2 tdev{1..5}
PMU2 tdie{1..10}

I have Mac Mini M4 base model (4 perf, 6 eff cores, 10 GPU).

exelban, 2024-12-05:
HID sensors valid only for some revisions of M1

EriksRemess, 2024-12-05:
Still temperatures reported by those HID sensors correlate with CPU load and feel to the touch. None of them are unrealistic like showing negative temperature and only tcal ones are static but those are probably for some calibration.

EriksRemess, 2024-12-05:
for SMC sensors - candidates I think would be TPD{0..F} ones. Attached smc tool outputs for under load and without load. Those temperatures are also same that PMU and PMU2 hid sensors report.
[load.txt](https://github.com/user-attachments/files/18023789/load.txt)
[unload.txt](https://github.com/user-attachments/files/18023791/unload.txt)

exelban, 2024-12-05:
Yes, TPD looks fine. But for me TpXX is more like CPU. Cause it goes much hotter than TPD. Also TpXX was in m2. I think more test cases are needed. And maybe with a different kind of load. Maybe GPU also, to gain info on what sensors are for GPU.

mo-corner-404, 2024-12-07:
I ran the script from this link for 3 different scenarios:
1. [idle.txt](https://github.com/user-attachments/files/18046577/idle.txt), doing nothing just ran the script for ~ 1 minute
2. [load.txt](https://github.com/user-attachments/files/18046578/load.txt), spawning multiple `yes > /dev/null & yes > /dev/null & yes > /dev/null & yes > /dev/null &` to try stress load the CPU
3. [gpu.txt](https://github.com/user-attachments/files/18046580/gpu.txt), stress load the GPU using Basemark GPU free with the custom test running @ 1920*1080 non fullscreen window

Running a MacBook Pro M4 10-Core (4 Perf + 6 Eff)
Hope it helps with the sensor mapping 😄 

> yes, basically it looks like Apple has changed the sensor keys again. If there is someone who wants to help I would recommend taking a look [here](https://github.com/exelban/stats/issues/1703) how it could be done.
> 
> Since I do not have an m4 mac I cannot help with that process.

EriksRemess, 2024-12-07:
While TpXX goes higher, variance seem to be bit too much (without load 12-20C between sensors) and unrealistic to be 100+ celsius at load - that's throttling territory.

exelban, 2024-12-09:
@mo-corner-404 thx, but unfortunately you collect only power sensors without any temperature sensors

Kencho5, 2024-12-10:
will this be fixed? i don't have percentages on m4 pro

exelban, 2024-12-10:
@Kencho5 you help to fix that? I don't have sensors keys in the app

nirbhaymalik, 2024-12-10:
Hello @exelban,
thank you for working on this application, I am sure hundreds of people use it.

Unfortunately, I do not have enough knowledge on the application to provide any assistance, but I would like to share a behaviour which seems different than what others are facing. From what I read above, the app is unable to read Sensors' and Power consumption information. 

In my case, all readings are available except the GPU temperatures. It may be possible that I am not aware about other stats that were working in previous versions. Please refer to the attached screenshot of the GPU settings. 
[image]

I can confirm that GPU consumption seems to be working, as it changes the reading after putting load on the machine.

I hope this information is useful, please reach out if you would like to know further details about it.

Looking forward to your fix for this issue.

Cheers

exelban, 2024-12-10:
@nirbhaymalik sorry, but do you read the issue?

nirbhaymalik, 2024-12-10:
@exelban I did, and is why I shared whatever information I could, as you requested from your users. My apologies if my shared information is of no use. More than happy to share anything else you may need, but I may require some instructions if you want me to run share stats straight from the terminal.

exelban, 2024-12-10:
the best you can do is to use the script and make a test with/without load. The script could be found in the issue I mention earlier.

PS: I will adjust the script to be easier to work with and will share a link here.

fran-6co, 2024-12-10:
I will also be happy to help. I did run the script after running the `Makefile` but I think it will be easy if it creates a output.txt with the output. I tried to, but couldn't, sorry. 

I'm on a base M4 Mac mini.

exelban, 2024-12-10:
I will prepare a python script as soon as possible to help with getting the data and share it here

Kencho5, 2024-12-10:
can you tag me when you prepare it? i will run it

trinquity, 2024-12-10:
I tried to monitor sensors however even with slight change with the script I wasn't able to print any temperature sensors at all. Need to re-write the part where it prints the sensors. 
I will be posting the results if you share the proper script @exelban

Sonnyger, 2024-12-11:
@exelban How can it be that there are already some apps that can correctly read the M4 sensors of all CPU and GPU cores? As an example, TG Pro. Can't you just "copy" that?

exelban, 2024-12-11:
First of all they can just by the new Mac and detect a new keys basically in first day of release a new silicon.
Or maybe they have some method to obtain that information from macOS that are not known publicly.

DigiH, 2024-12-11:
Stop rushing him 😛 it's not like your M4 Mac is going up in flames if there is no full sensors temperatures display in Stats for a while.

I think @exelban is doing what he can to get these sensors in correctly eventually, without him having an M4, like preparing a script for us with M4s to ru.

For anyone else with an immediate need for these - this is open source software, so you can feel free to get the qualification to implement this yourselves and just submit a PR.

exelban, 2024-12-11:
Ok, so basically this script helps a lot with finding a sensors. It shows a changes on sensors over the time. I wanted to add similar functionality to the app but have no time for that. 

[Link](https://gist.github.com/exelban/ec24807e0fd4326b1bca962e8ae5786a)

You can use it by downloading and running: 
`(while true; do /Applications/Stats.app/Contents/Resources/smc list -p; sleep 1; done) | python3 watch-smc.py`

Dorkington, 2024-12-11:
I have an M4 Max and am willing to send you whatever results from the script, but I'm also kind of a code dummy. 

I ran `pip install colorama` to install colorama, then ran your above code but get the error 

```
Traceback (most recent call last):
  File "/Users/jason/Downloads/ec24807e0fd4326b1bca962e8ae5786a-393b769a1143cdf9977718e20bb8438cdd81675b/watch-smc.py", line 6, in <module>
    from colorama import Fore, Back, Style
ModuleNotFoundError: No module named 'colorama'
```

As far as I can see, colorama is installed though, so I'm a bit lost sorry

exelban, 2024-12-11:
most probably you need to install colorama with pip3 not pip

Dorkington, 2024-12-11:
Ended up having to do pip3 install colorama --break-system-packages after pip3 shot out a bunch of errors... hopefully I didn't ruin anything

Does this script run until manually stopped, or does it have a stop point?

exelban, 2024-12-11:
manually

trinquity, 2024-12-11:
[m4-p-smc.txt](https://github.com/user-attachments/files/18103089/m4-p-smc.txt)
I only got this from the script @exelban cpu was on stress by geekbench but i think they are still the wattage sensors

exelban, 2024-12-11:
oh, sorry. Use this command: 
`(while true; do /Applications/Stats.app/Contents/Resources/smc list -t; sleep 1; done) | python3 watch-smc.py`

trinquity, 2024-12-11:
thank you @exelban , got captured some sensor data tested with geekbench 6 GPU and CPU tests.

[m4-idle-cpu.txt](https://github.com/user-attachments/files/18103360/m4-idle-cpu.txt)
[m4-idle-gpu.txt](https://github.com/user-attachments/files/18103361/m4-idle-gpu.txt)

exelban, 2024-12-11:
I have removed some other keys from the script. Please try again. For now lets find a CPU, omit the GPU for now.

Sonnyger, 2024-12-12:
With this data you can find out which temperatures belong to which cores of the CPU and GPU? I wish I had skills like that, really remarkable.

fran-6co, 2024-12-12:
I also run the script. 
One file is idle, no load on any app.

- [BaseM4Macmini_Mac16,10_4P+6E_16GB_idle.txt](https://github.com/user-attachments/files/18111002/BaseM4Macmini_Mac16.10_4P%2B6E_16GB_idle.txt)

The other is running a _ffmpeg_ conversion with _libx265_ which uses only CPU. Fan on my mac mini was at 20% (all time is at that speed), but after 70 seconds of running the script I forced it to 100% to see if any decrease of temperature.

- [BaseM4Macmini_Mac16,10_4P+6E_16GB_ffmpeg_libx265.txt](https://github.com/user-attachments/files/18111001/BaseM4Macmini_Mac16.10_4P%2B6E_16GB_ffmpeg_libx265.txt)

Hope it helps.

nirbhaymalik, 2024-12-13:
Hey @exelban, I ran the script on M4 Pro chip on idle condition.
Check out the text files below, I have also included a file with no colour codes for readability.

[m4_pro_idle_no_colour.txt](https://github.com/user-attachments/files/18121308/m4_pro_idle_no_colour.txt)
[m4_pro_idle.txt](https://github.com/user-attachments/files/18121309/m4_pro_idle.txt)

Kencho5, 2024-12-18:
any updates on this?

DevLiuSir, 2024-12-21:
Is there any new progress? There is no CPU temperature sensor list for m4max and pro

mo-corner-404, 2024-12-23:
Run the script:
1. Idle for 1 minute [idle.txt](https://github.com/user-attachments/files/18229437/idle.txt)
2. Idle for 10 seconds then max load all CPU cores for 1 minute [stress.txt](https://github.com/user-attachments/files/18229438/stress.txt)

trinquity, 2024-12-23:
@Kencho5 its still on, in-progress. I believe @exelban working on it, mapping sensors are kinda hard on new systems. You can follow the status of the issues by [here](https://github.com/users/exelban/projects/1/views/2)

Strapples, 2025-01-05:
Subscribed to this thread. I’ll see what I can do if anything I can test new builds and run scripts for the devs to test stuff. I myself have VSCode and know enough to find bugs and debug some stuff. Mostly know my way around NodeJS. Swift is something I wanna get into messing with so may as well pull your code onto my machine and start tinkering with it and see if I can get anything to behave. Particularly that value that’s pegged to the upper limit lulz

lemme know where to start :)

exelban, 2025-01-30:
Hi. Please try this build:

[Stats.dmg.zip](https://github.com/user-attachments/files/18605178/Stats.dmg.zip)

fran-6co, 2025-01-30:
Do you want us to share the output of the script or just test it to see the sensors?

From my side, I see inconsistency on the quantity of cores. I have base M4 and it has 4 performance and 6 efficiency, but it shows 6+6.

GPU has 10 cores, and it shows 8.

The picture is with no load at all.
[image]

exelban, 2025-01-30:
yeap, looks ok. I'm more interested in another cpu rather than m4.

IsmaelP19, 2025-01-30:
This is what I get on my m4 pro (8 performance + 4 efficiency cores) with 16 cores of gpu

[image]

josephjorgensen, 2025-01-30:
I have an M4 Pro 14 core (10+4), my smc list is attached (in case another is helpful).

[smc-list-m4pro.txt](https://github.com/user-attachments/files/18607193/smc-list-m4pro.txt)

IsmaelP19, 2025-01-31:
> [@IsmaelP19](https://github.com/IsmaelP19) could you please share the output of `/Applications/Stats.app/Contents/Resources/smc list`? (as a file please)

Sorry, but I'm a bit lost.

I downloaded your script from [here](https://gist.github.com/exelban/ec24807e0fd4326b1bca962e8ae5786a).
After that, I installed Colorama with pip3 and ran `(while true; do /Applications/Stats.app/Contents/Resources/smc list -t; sleep 1; done) | python3 watch-smc.py`.

I stopped the script after a few minutes and got this error:
```
^CTraceback (most recent call last):
  File "/Users/ismaelperez/Developer/watch-smc.py", line 33, in <module>
    for line in fileinput.input():
                ~~~~~~~~~~~~~~~^^
  File "/Library/Frameworks/Python.framework/Versions/3.13/lib/python3.13/fileinput.py", line 251, in __next__
    line = self._readline()
KeyboardInterrupt
```
Anyone help me with getting the smc list?
I'll wait for further instructions

exelban, 2025-01-31:
@IsmaelP19 I was just asked output of smc, not running a script. But anyway, @josephjorgensen already shared everything I need.

exelban, 2025-01-31:
thx, I already have from m4 Pro. Please try this build:

[Stats.dmg.zip](https://github.com/user-attachments/files/18615719/Stats.dmg.zip)

IsmaelP19, 2025-01-31:
> thx, I already have from m4 Pro. Please try this build:
> 
> [Stats.dmg.zip](https://github.com/user-attachments/files/18615719/Stats.dmg.zip)

I see 2 missing efficiency cores and other 8 GPU cores
[image]

exelban, 2025-01-31:
nope, everything looks ok. It's not a sensor per core. It's just a sensors on a die. It has no correlation with cores number.

closed 2025-02-02 by exelban