exelban/stats #2250: Can't update from within the app

teiffs, 2024-12-02:
**Describe the bug**
running into same (slightly) "type" of issue as #1081. I am using a "standard" (non Admin) user account. 

Spent a couple of months ignoring the issue and just NOT updating until today. What happens is when I hit the install button after the download it reports "no permissions on /Application". So it won't update. Today after the failed attempt I shut down Stats and then opened the downloaded dmg and moved it into Applications as if a new installation (replace existing).

**Details:**
 - Device: [iMac 2017]
 - macOS: [13.7.1]
 - Application version: [2.11.19] I am current at the moment.

iamsoum, 2024-12-19:
Confusing issue.

> Today after the failed attempt I shut down Stats and then opened the downloaded dmg and moved it into Applications as if a new installation (replace existing).

So did that work? Did it update the app without losing the settings you customized in the previous version on your Mac?

@exelban it looks like the issue is that there's no update button in the app settings, I came here looking for a similar issue before I opened my own.

exelban, 2024-12-19:
@iamsoum wat?
It's issue related only to the update when account have no permission to the Application folder.

theRealKLH, 2024-12-20:
> Confusing issue.
> 
> So did that work? Did it update the app without losing the settings you customized in the previous version on your Mac?
> 
@iamsoum yes. it did. There is a "check for updates" button on the settings tab.

with stats closed I was able to move from the dmg to the applications folder and it prompted with the "keep both, stop, replace". 

@exelban additional info when manually using the downloaded .dmg
With Stats open..
[image]

With Stats closed it prompts to enter admin creds and proceeds (if you enter valid info of course). This prompt doesn't happen when using the internal "upgrade" mechanism. Instead giving the "no permissions" error.

exelban, 2024-12-20:
@theRealKLH so everything works fine. don't see any issues in what you described

theRealKLH, 2024-12-20:
Well actually I'm describing your update mechanism isn't "functioning" as a user would expect. Your upgrade has a "download" button which then turns into an "install" button. After progress hits 100%, clicking install doesn't work AND you can't close Stats else the "install" dialogue disappears. Which then the user would have to find the DMG and do the "drag to Applications".

xa4hf8, 2025-01-05:
Should it prompt for privilege escalation?

[image]

Maybe the [Sparkle framework](https://sparkle-project.org) represents a solution?

closed 2025-01-26 by exelban