exelban/stats #2251: Combined modules setting makes modules disappear

atakdd, 2024-12-03:
This may not be a bug, I normally don't use this setting except only on my mac desktop however I just finished setting up a new m3 macbook air and the display on the air has a cutout in the center of the very top as the display is edge to edge. Because of the cutout, I have limited space as far as how many modules I can physically see, at the moment I can only view one. I wanted to combine all the modules so that I can use one module that shows cpu, gpu, etc. however when I select the Combined Modules option in the settings, the singular module that was in the top bar just disappeared. I don't see any additional settings in the app to try and adjust it's visibility, so I am wondering if I have found a bug for this specific macbook given the display.

closed 2024-12-03 by exelban

atakdd, 2024-12-03:
If what I am experiencing is not a bug, is there a feature in which I can accomplish what I am trying to do? Combined modules just hides any icon.

exelban, 2024-12-03:
cause there is no space for it to fit in the menu bar.

atakdd, 2024-12-03:
Then how come when I disable the Combined Modules, the first icon in the list does show up? I would have thought given there is enough space to show 1 module, combining everything into 1 module would not have been an issue.

Deniz

Sent with [Proton Mail](https://proton.me/mail/home) secure email.