exelban/stats #2260: stats disk monitoring blocks IO for interfacing with S.M.A.R.T data

vaibhavBh-0, 2024-12-05:
Description:
If stats is running and is monitoring the disk, smartmontools cli `smartctl -A disk0` does not work and yields the following output.

[image]

If I stop the disk monitoring and restart the app. smartctl is able to read the S.M.A.R.T and report the data properly.
I observed the same behavior with DriveDx.

Details:
- Device [Macbook Pro 2020 (13-inch, 2020, Four Thunderbolt 3 ports)]
- macOS [15.1.1 (24B91)]
- Application version [2.11.19]

exelban, 2024-12-06:
yes, smart driver could be used once per time. You need to close stats or disable the disk module.

closed 2024-12-06 by exelban

vaibhavBh-0, 2024-12-06:
This was not an issue earlier (before I updated to macOS 14.7). I could run the smartctl command and use the stats disk monitoring.

exelban, 2024-12-06:
I have never use another tool if Stats is running. There was no changes in SMART logic in a months.