exelban/stats #2262: does not recognize all drives

wotography, 2024-12-07:
I have two SSD T5 (Samsung) connected with same amount of storage. the stats only show one of them.

wotography, 2024-12-17:
Okay, so you don't care about it? Or want to have a ceal issue list on github?

exelban, 2024-12-17:
share please the output of `diskutil list` with this specific ssd

wotography, 2024-12-17:
```
/dev/disk4 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *250.1 GB   disk4
   1:                  Apple_HFS WW002                   250.1 GB   disk4s2

/dev/disk5 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *250.1 GB   disk5
   1:                  Apple_HFS WW003                   250.1 GB   disk5s2
```

exelban, 2024-12-17:
check please the removable option in the disk module settings

wotography, 2024-12-18:
its turned on. and i found out it is available in the dropdown list. but its not shown in the overview. see the screenshots.

only one of them is displayed in the menu overview

[image]
[image]
[image]

exelban, 2024-12-18:
I will make a debug build that will help to figure out what is going on here

exelban, 2024-12-18:
could you check if all disks are visible in the Popup settings tab?

wotography, 2024-12-19:
its just the one drive there.
[image]

exelban, 2024-12-19:
could you please share the output of:
`diskutil info /dev/disk4s1 | egrep -om1 "([A-Z0-9]{8}.[A-Z0-9]{4}.+$)"`
`diskutil info /dev/disk5s1 | egrep -om1 "([A-Z0-9]{8}.[A-Z0-9]{4}.+$)"`

wotography, 2024-12-19:
for both:
00000000-0000-0000-0000-000000000000

Not sure why actually. I use both the ssds with my pocket cinema camera from blackmagic design and usualy format it with the camera.

exelban, 2024-12-19:
wow, just to ensure could you check that for both disks?
Disk Utility → Select Disk/Partition → click "Info"

And check UUID field

exelban, 2024-12-19:
looks like something is broken in the disk drivers, since disk has basically no uuid. I will try to make workaround with that...

exelban, 2024-12-19:
try this build please:
[Stats.dmg.zip](https://github.com/user-attachments/files/18196844/Stats.dmg.zip)

wotography, 2024-12-19:
works fine.
by overwriting my original install, did i just delete my settings with it?

[image]

closed 2024-12-22 by exelban

wotography, 2024-12-22:
Fantastic! Thank you very much! :)