exelban/stats #2267: 无故退出

hughzone, 2024-12-13:
设备：Mac mini(M1) 16+512 macOs Sequoia 15.1.1
问题描述：
1.软件在运行一段时间后会自动退出，需手动再次打开软件，此现象在高内存负荷时更易出现；
2.系统休眠后再次启动电脑软件也出现已退出运行的情况。

qkdxorjs1002, 2024-12-17:
I'm also constantly experiencing this problem.

### Crash reports:

```plain
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [665]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.11.20 (639)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-12-10 23:05:27.4546 +0900
OS Version:            macOS 15.1.1 (24B2091)
Report Version:        12
Anonymous UUID:        C054AC01-0F40-E0F5-A96E-A2421B4D855C

Sleep/Wake UUID:       DCA263FD-6EC8-41E3-9EBE-4F4AA16C84D7

Time Awake Since Boot: 54000 seconds
Time Since Wake:       10577 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x0000000195cc7f54

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [665]

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libswiftCore.dylib            	       0x195cc7f54 swift_unknownObjectRetain + 44
1   libswiftCore.dylib            	       0x19597b95c String._bridgeToObjectiveCImpl() + 92
2   Kit                           	       0x100e58820 0x100da8000 + 722976
3   Kit                           	       0x100e52b7c 0x100da8000 + 699260
4   Kit                           	       0x100e52cb8 0x100da8000 + 699576
5   AppKit                        	       0x188948014 _NSViewDrawRect + 160
6   AppKit                        	       0x1892ec9fc -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 1084
7   AppKit                        	       0x188947a08 -[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:] + 536
8   AppKit                        	       0x188faeca8 -[NSViewBackingLayer drawInContext:] + 56
9   AppKit                        	       0x188c3f31c 0x1887f9000 + 4481820
10  AppKit                        	       0x188c3ae40 0x1887f9000 + 4464192
11  AppKit                        	       0x188c36698 0x1887f9000 + 4445848
12  AppKit                        	       0x188c3aa20 0x1887f9000 + 4463136
13  AppKit                        	       0x188c38590 0x1887f9000 + 4453776
14  AppKit                        	       0x188cacab4 0x1887f9000 + 4930228
15  AppKit                        	       0x188cace64 0x1887f9000 + 4931172
16  AppKit                        	       0x188fae8d4 -[NSViewBackingLayer display] + 1272
17  QuartzCore                    	       0x18d7d9a40 -[CALayer renderInContext:] + 140
18  QuartzCore                    	       0x18d7db6dc -[CALayer _renderSublayersInContext:] + 340
19  QuartzCore                    	       0x18d7d9e78 -[CALayer renderInContext:] + 1220
20  QuartzCore                    	       0x18d7db6dc -[CALayer _renderSublayersInContext:] + 340
21  QuartzCore                    	       0x18d7d9e78 -[CALayer renderInContext:] + 1220
22  QuartzCore                    	       0x18d7db6dc -[CALayer _renderSublayersInContext:] + 340
23  QuartzCore                    	       0x18d7d9e78 -[CALayer renderInContext:] + 1220
24  QuartzCore                    	       0x18d7db6dc -[CALayer _renderSublayersInContext:] + 340
25  QuartzCore                    	       0x18d7d9e78 -[CALayer renderInContext:] + 1220
26  AppKit                        	       0x1892ec108 -[NSView _renderLayerInContext:] + 204
27  AppKit                        	       0x1892ec7c8 -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 520
28  AppKit                        	       0x1892e6920 __47-[NSView displayRectIgnoringOpacity:inContext:]_block_invoke + 656
29  AppKit                        	       0x1888755e0 NSPerformVisuallyAtomicChange + 108
30  AppKit                        	       0x188a76574 -[NSView displayRectIgnoringOpacity:inContext:] + 104
31  AppKit                        	       0x188a764bc __62-[NSView cacheDisplayInRect:toBitmapImageRep:includeSubviews:]_block_invoke + 192
[... 458 lines cut ...]
__CTF                               824        1 
__DATA                            23.8M      789 
__DATA_CONST                      22.0M      816 
__DATA_DIRTY                      2249K      281 
__FONT_DATA                        2352        1 
__GLSLBUILTINS                    5174K        1 
__INFO_FILTER                         8        1 
__LINKEDIT                       588.7M       16 
__OBJC_RW                         2355K        1 
__TEXT                           975.2M      836 
__TPRO_CONST                       272K        2 
libnetwork                         128K        8 
mapped file                      328.8M       73 
owned unmapped memory             9584K        1 
page table in kernel               801K        1 
shared memory                     1600K       28 
===========                     =======  ======= 
TOTAL                              4.2G     5217 
TOTAL, minus reserved VM space     4.1G     5217 
.
.
.
```

[Stats_reports.zip](https://github.com/user-attachments/files/18166290/Stats_reports.zip)

exelban, 2024-12-17:
could you try to disable Network module?

qkdxorjs1002, 2024-12-17:
The network module is already turned off.

[video](https://github.com/user-attachments/assets/9a2819a3-a58a-49a9-ac0e-7e88a7119f59)

qkdxorjs1002, 2024-12-17:
New Report

```plain
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [14863]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.11.21 (641)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-12-17 23:12:31.1858 +0900
OS Version:            macOS 15.1.1 (24B2091)
Report Version:        12
Anonymous UUID:        C054AC01-0F40-E0F5-A96E-A2421B4D855C

Sleep/Wake UUID:       7B9046CC-21FB-434B-ADA4-01137A5EF390

Time Awake Since Boot: 21000 seconds
Time Since Wake:       2510 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000046
Exception Codes:       0x0000000000000001, 0x0000000000000046

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [14863]

VM Region Info: 0x46 is not in any region.  Bytes before following region: 4311187386
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      100f78000-100fa8000    [  192K] r-x/r-x SM=COW  /Applications/Stats.app/Contents/MacOS/Stats

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libswiftCore.dylib            	       0x1981922e4 swift::runtime::AccessSet::insert(swift::runtime::Access*, void*, void*, swift::ExclusivityFlags) + 64
1   libswiftCore.dylib            	       0x198192514 swift_beginAccess + 84
2   AppKit                        	       0x18b147964 0x18ad31000 + 4286820
3   AppKit                        	       0x18ae76384 -[NSWindow _copyNeedsDisplayRegionInRect:validateSubtractedRegion:] + 112
4   AppKit                        	       0x18b81bcd4 __80-[NSView _oldDisplayRectIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:]_block_invoke + 256
5   AppKit                        	       0x18adad5e0 NSPerformVisuallyAtomicChange + 108
6   AppKit                        	       0x18ae76120 -[NSView _oldDisplayRectIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:] + 108
7   AppKit                        	       0x18adad5e0 NSPerformVisuallyAtomicChange + 108
8   AppKit                        	       0x18ae760a8 -[NSView _displayRectIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:] + 108
9   AppKit                        	       0x18b81ddb0 __25-[NSView displayIfNeeded]_block_invoke + 448
10  AppKit                        	       0x18adad5e0 NSPerformVisuallyAtomicChange + 108
11  AppKit                        	       0x18ae1279c -[NSView displayIfNeeded] + 192
12  AppKit                        	       0x18ae0fb40 -[NSWindow displayIfNeeded] + 264
13  AppKit                        	       0x18ae0f994 __NSWindowGetDisplayCycleObserverForDisplay_block_invoke + 520
14  AppKit                        	       0x18ae0e8c4 NSDisplayCycleObserverInvoke + 168
15  AppKit                        	       0x18ae0e524 NSDisplayCycleFlush + 672
16  QuartzCore                    	       0x18fca19d8 CA::Transaction::run_commit_handlers(CATransactionPhase) + 120
17  QuartzCore                    	       0x18fca0740 CA::Transaction::commit() + 320
18  AppKit                        	       0x18ae8f334 __62+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayLink]_block_invoke + 272
19  AppKit                        	       0x18b86676c ___NSRunLoopObserverCreateWithHandler_block_invoke + 64
20  CoreFoundation                	       0x187241078 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
21  CoreFoundation                	       0x187240f64 __CFRunLoopDoObservers + 552
22  CoreFoundation                	       0x187240594 __CFRunLoopRun + 788
23  CoreFoundation                	       0x18723fbc4 CFRunLoopRunSpecific + 588
24  HIToolbox                     	       0x1926b1f64 RunCurrentEventLoopInMode + 292
25  HIToolbox                     	       0x1926b7bb4 ReceiveNextEventCommon + 220
26  HIToolbox                     	       0x1926b7eb8 _BlockUntilNextEventMatchingListInModeWithFilter + 76
[... 559 lines cut ...]
  "logWritingSignature" : "83a69753a4a98787125cb0938b182a170cf85332",
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "5fb4245a1bbfe8005e33a1e1",
      "factorPackIds" : {

      },
      "deploymentId" : 240000021
    },
    {
      "rolloutId" : "63508950b3714d3622fc77f7",
      "factorPackIds" : {
        "SIRI_MEMORY_SYNC_CONFIG" : "6526e111d0c9ce2f459b54a8"
      },
      "deploymentId" : 240000015
    }
  ],
  "experiments" : [

  ]
}
}

```

exelban, 2024-12-17:
disable combined modules option

qkdxorjs1002, 2024-12-17:
Okay, I'll wait and see if there's a crash again.

hughzone, 2024-12-23:
> 禁用组合模块选项

Let me use this method for a period of time to evaluate whether the issue still persists.

hughzone, 2024-12-23:
> disable combined modules option

Let me use this method for a period of time to evaluate whether the issue still persists.

closed 2025-01-26 by exelban

bzadon, 2025-02-17:
reporting that i have combined modules enabled and seeing this issue on a daily basis. don't have a crash log yet. this is a new issue that hasn't been seen before. TIA!