exelban/stats #2285: New v2.11.22 efficiency and performance cores frequencies don't look right for Mac mini M4

DigiH, 2024-12-22:
New v2.11.22 efficiency and performance cores frequencies don't look right for Mac mini M4

**Details:**
 - Device: [Mac mini M4
 - macOS: 15.2
 - Application version: 2.11.22

[image]

Power efficient, yes, but not that efficient 😉

SDpower, 2025-01-01:
Check this.
https://github.com/fastfetch-cli/fastfetch/discussions/1394

exelban, 2025-01-04:
Hi. Could someone with m4 run this build for a few seconds with opened popup view?
[Stats.dmg.zip](https://github.com/user-attachments/files/18307224/Stats.dmg.zip)

It will collect some logs to the log.txt that could be find in the Documents folder. And share the logs please.

andrewr-dev, 2025-01-04:
Mac mini M4 Pro
[log.txt](https://github.com/user-attachments/files/18308176/log.txt)

exelban, 2025-01-04:
thx, please check if there are any output of:
`ioreg -r -f -x -c AppleARMIODevice | grep "voltage-states1-sram"`
and
`ioreg -r -f -x -c AppleARMIODevice | grep "voltage-states5-sram"`

andrewr-dev, 2025-01-04:
```
  |   "voltage-states1-sram" = 
  |     00000000: 60 90 0F 00 16 03 00 00 60 6C 15 00 20 03 00 00 60 48 1B 00 3E 03 00 00 00 3A 20 00 5C 03 00 00 `.......`l.. ...`H..>....: .\...
  |     00000020: 80 E3 23 00 84 03 00 00 A0 A2 26 00 AC 03 00 00 00 8D 27 00 AC 03 00 00                         ..#.......&.......'.....

  |   "voltage-states5-sram" = 
  |     00000000: E0 39 13 00 16 03 00 00 40 12 17 00 2A 03 00 00 40 77 1B 00 43 03 00 00 40 DC 1F 00 5C 03 00 00 .9......@...*...@w..C...@...\...
  |     00000020: 80 E3 23 00 75 03 00 00 C0 EA 27 00 89 03 00 00 20 C3 2B 00 98 03 00 00 C0 3D 2F 00 AC 03 00 00 ..#.u.....'..... .+......=/.....
  |     00000040: A0 5A 32 00 C0 03 00 00 E0 EA 34 00 CA 03 00 00 40 4C 37 00 E8 03 00 00 E0 4F 39 00 FC 03 00 00 .Z2.......4.....@L7......O9.....
  |     00000060: E0 C6 3A 00 FC 03 00 00 20 E0 3B 00 FC 03 00 00 60 F9 3C 00 FC 03 00 00 E0 B4 3D 00 FC 03 00 00 ..:..... .;.....`.<.......=.....
  |     00000080: 40 9F 3E 00 FC 03 00 00 00 62 43 00 38 04 00 00 00 D9 44 00 38 04 00 00                         @.>......bC.8.....D.8...

```

exelban, 2025-01-06:
hi, please try this build:
[Stats.dmg.zip](https://github.com/user-attachments/files/18321437/Stats.dmg.zip)

andrewr-dev, 2025-01-07:
I don't see any log.txt with the new build. 
Efficiency and Performance core frequencies still show as 1MHz

exelban, 2025-01-08:
what about this one:

[Stats.dmg.zip](https://github.com/user-attachments/files/18350082/Stats.dmg.zip)

andrewr-dev, 2025-01-08:
Looks better, frequencies are in the correct range.

Minimal CPU usage:
[image]

High CPU usage:
[image]

closed 2025-01-12 by exelban