exelban/stats #2287: Feature Request: Display Data in App directly

nicolopadovan, 2024-12-23:
I have been using Stats for quite some time, and it helps me quite a lot in monitoring the resource usage of my Mac.
I mostly work from a work station with two external big monitors, so I never paid attention to the fact that the data can only be accessed via Menu Bar Widgets and, limited to CPU, GPU, Disk, RAM and Network, Control Center Widgets.
Having to sometimes switch to using just my laptop, I have noticed that being able to monitor the data from the Stats App GUI as well would be highly beneficial for my use case.
I will be working on a PR to implement this myself as well, since I do think that this is not at all urgent or in the current Stats roadmap, yet it could be a helpful little View to have within the app.

@exelban do let me know what you think of this.

exelban, 2025-01-06:
Hi. Thanks for your feedback. I'm thinking about that for some time. And pretty sure it will be added. But now I have no time for that such as an idea of how it could looks like.

zhihui1999, 2026-01-21:
I think making a simple interface that displays all information by column is enough