exelban/stats #2288: UI problem in CPU section

TeoConan, 2024-12-25:
**Description**
Texts are superimposed in CPU section, I don't know why this happened

[image]

**Details:**
 - Device: Macbook Air 2020 M1
 - macOS: Sequoia 15.1.1 (24B91)
 - Application version: 2.11.22

TeoConan, 2024-12-27:
[image]

Nevermind, after a reboot, everythings back to normal
Strange !

exelban, 2024-12-27:
@TeoConan I have noticed that there is a bug with titles that are too long. Please open an issue with that cause I can forget about that.

closed 2025-01-05 by exelban