exelban/stats #2299: Sensor energy consumption display is abnormal

l1kem, 2025-01-03:
As shown in the picture, the energy consumption perceived by the sensor was abnormal. It returned to normal after the software was restarted, but it still showed abnormal after a period of time.
[image]

Software Version: 2.11.22
Computer-related Information:
Model: MacBook Pro 16 2021 M1 Max
System: macOS Sequoia 15.2

exelban, 2025-01-06:
hi. Please run this build (it has only Sensors module) till the total consumption will be strange. This build will collect the logs into log.txt that could be find in the Documents folder. And share the log after please.

[Stats.dmg.zip](https://github.com/user-attachments/files/18322177/Stats.dmg.zip)

POEG1726, 2025-01-13:
Same problems occurs on my laptop since updated to MacOS15, it seems like all fan control app doesn't work anymore.

closed 2025-01-19 by exelban