exelban/stats #2319: Add calendar events to the clock popup (feature request)

AlbatorLaho, 2025-01-09:
I recently found out that this app has a clock widget, and that it can show the calendar! (pretty neat actually)

Anyway, it seems to do almost everything I'd like—coming form Itsycal—except it can't show calendar events... (and a few other minor things)

Any chance that this would/could be added? It seems kind of weird having an app that's mostly associated with system status info, having a calendar... but it does sort of make sense!

R-S-T, 2025-01-11:
would like to add a +1 to showing calendar events. That would be awesome!

exelban, 2025-01-12:
hi. No there is no plans for events in the calendar. Sorry, but it's a monitoring app, not a calendar app. I have no enough time to support a calendar features as well. There are a lot of good calendars for macOS.

closed 2025-01-12 by exelban