exelban/stats #2321: Accessibility bug: Stats menu icons are not read well by VoiceOver

pitermach, 2025-01-10:
Hello. I found Stats after I started looking for alternatives to iStatMenus. I’m blind and use the VoiceOver feature of Mac OS, which reads everything displayed on the screen and lets me navigate through and interact with the user interface with just keyboard commands. When iStatMenus 7 came out the app redesign made that app completely unusable with VoiceOver, prompting me to look for alternatives. Over all, stats works great with VoiceOver, both when you click on one of the modules to read its details or go into the settings to make adjustments. There are a few things that could make the app even better for blind users. One is implementing keyboard shortcuts to quickly open modules, which is already mentioned in #1976 so I’d just like to add my voice to that. ☺️

The next issue is what this report is about. Right now the status menu items for the different stats modules don’t have good accessibility labels, which makes it very annoying for a VoiceOver user to know which module you’re focused on.

To reproduce:

- Press Command+F5 to turn on VoiceOver. You can press this shortcut again to turn it off when you want it to stop talking.
- Press CTRL+Option+M twice to move the cursor to the status menus section of the menu bar.
- Press CTRL+Option+Right and left to navigate between the menu bar items.

Notice that the system icons, like the one for Wi-Fi or the clock, announce what they display, like the current time or signal strength for wi-fi. When iStatMenus still worked, its icons also announced the displayed information - IE the CPU icon read the usage, the network icon announced the download/upload speed and so on.

In contrast, VoiceOver speaks every icon of Stats as “Stats status menu, so if I want to find a specific icon I have to activate all of them to see which item corresponds to what, and then memories this for the future and hope that Mac OS doesn’t decide to shift the icon order around.

It would be great if the stats icons got clear labels, at a minimum announcing what module they are for but better if they would read the information being displayed.

Device info: Mac OS 15.2 with stats 2.11.23 on M1 and M3 pro Macs.

exelban, 2025-01-12:
Hi. Thanks for your feedback. I will try to find time to add these improvements as soon as possible.

closed 2025-01-19 by exelban

pitermach, 2025-01-19:
The menu icons read much nicer now, thanks! Stats now has a permanent spot on my menu bar :)

exelban, 2025-01-19:
@pitermach happy to help. I have tried to go throw the app adding missing tooltips and enabling the missing accessibility blocks. But maybe I miss something.

In case something must be added open a new PR or just ping me in any way. I have in mind to add the module value to the widget. But it will take some time.

pitermach, 2025-01-19:
Will do if I find anything else. Yeah, reading the value on the menu icon will definitely be great if you can implement it in the future, but at least with this update I know what I'm clicking on which was the most important problem.

Rogue-Git-Dev, 2025-01-21:
I confirm the labels for each module are spoken by VoiceOver. The values are not [still mini, speed, battery, stack, et cetera] but this is a huge improvement. For a long time I have used Stats but now it is so much nicer. I anticipate more VoiceOver related improvements.