exelban/stats #2329: Displayed core frequencies

vdhamer, 2025-01-11:
This is on an M1 Ultra during heavy compute load (export of 500 photos in Lightroom Classic).
The number of 600 MHz stays the same during idling and during heavy workload.
I don't expect it to be that low. And expect (unsure) that the frequency can auto adjust depending on work load.

Stats 2.11.23 (currently latest version).

[image]

exelban, 2025-01-11:
looks like it does not calculate correctly for ultra. Does it change at all or always stays at 600?
I have also noticed the strange arrangement of load per core....hmmm

exelban, 2025-01-11:
hmm, please you that build and share the logs: https://github.com/exelban/stats/issues/2285#issuecomment-2571324324

vdhamer, 2025-01-11:
> looks like it does not calculate correctly for ultra. Does it change at all or always stays at 600? 

Have never seen a different value than 600 MHz.

> I have also noticed the strange arrangement of load per core....hmmm

The distribution of workload across cores… I haven’t checked it against say Apple’s utility (tricky because it constantly changes over time). But I think it is ok. This workload is one single application exporting (converting) many files by distributing the work over many cores. Probably the worker threads go to Performance Cores, while the coordination is done by a thread running on Efficiency Cores. Somehow I see (on both MacOS, but also Windows) that a single threaded load is spread out across multiple Efficiency cores. This is probably (if we forget the other Performance cores) to spread the thermal load better across the surface are of the chip: better to have 4 cores (2 per physical M1 Max chip) dissipate 25% of the energy on average than to have 1 of the force dissipate 100% of the energy. Again: not important for this use case where most of the power goes to Performance Cores. And I still think the shown graph and bars in your app are ok.

If really needed to figure out what’s going on, I can do testing on the M1 Ultra if you spec what to test.

vdhamer, 2025-01-11:
> hmm, please you that build and share the logs: [#2285 (comment)](https://github.com/exelban/stats/issues/2285#issuecomment-2571324324)

Where do I find the logs? Path or link to instructions?

exelban, 2025-01-11:
you can find logs in the Documents folder

vdhamer, 2025-01-11:
> you can find logs in the Documents folder

Ok. Hmmm. File date is May 19th 2024. So probably not what you are looking for.

----------------
2024-05-15 10:09:26 CombinedView.swift:229 DBG modules: 5
2024-05-15 10:09:26 CombinedView.swift:232 DBG adding: CPU portal
2024-05-15 10:09:26 CombinedView.swift:232 DBG adding: GPU portal
2024-05-15 10:09:26 CombinedView.swift:232 DBG adding: RAM portal
2024-05-15 10:09:26 CombinedView.swift:232 DBG adding: Disk portal
2024-05-15 10:09:26 CombinedView.swift:232 DBG adding: Network portal
----------------
2024-05-15 10:25:45 CombinedView.swift:229 DBG modules: 5
2024-05-15 10:25:45 CombinedView.swift:232 DBG adding: CPU portal
2024-05-15 10:25:45 CombinedView.swift:232 DBG adding: GPU portal
2024-05-15 10:25:45 CombinedView.swift:232 DBG adding: RAM portal
2024-05-15 10:25:45 CombinedView.swift:232 DBG adding: Disk portal
2024-05-15 10:25:45 CombinedView.swift:232 DBG adding: Network portal
----------------
2024-05-17 19:09:39 CombinedView.swift:229 DBG modules: 5
2024-05-17 19:09:39 CombinedView.swift:232 DBG adding: CPU portal
2024-05-17 19:09:39 CombinedView.swift:232 DBG adding: GPU portal
2024-05-17 19:09:39 CombinedView.swift:232 DBG adding: RAM portal
2024-05-17 19:09:39 CombinedView.swift:232 DBG adding: Disk portal
2024-05-17 19:09:39 CombinedView.swift:232 DBG adding: Network portal
----------------
2024-05-18 14:33:57 CombinedView.swift:229 DBG modules: 5
2024-05-18 14:33:57 CombinedView.swift:232 DBG adding: CPU portal
2024-05-18 14:33:57 CombinedView.swift:232 DBG adding: GPU portal
2024-05-18 14:33:57 CombinedView.swift:232 DBG adding: RAM portal
2024-05-18 14:33:57 CombinedView.swift:232 DBG adding: Disk portal
2024-05-18 14:33:57 CombinedView.swift:232 DBG adding: Network portal
----------------
2024-05-19 10:16:02 CombinedView.swift:229 DBG modules: 5
2024-05-19 10:16:02 CombinedView.swift:232 DBG adding: CPU portal
2024-05-19 10:16:02 CombinedView.swift:232 DBG adding: GPU portal
2024-05-19 10:16:02 CombinedView.swift:232 DBG adding: RAM portal
2024-05-19 10:16:02 CombinedView.swift:232 DBG adding: Disk portal
2024-05-19 10:16:02 CombinedView.swift:232 DBG adding: Network portal
----------------
2024-05-19 17:56:33 CombinedView.swift:229 DBG modules: 5
2024-05-19 17:56:33 CombinedView.swift:232 DBG adding: CPU portal
2024-05-19 17:56:33 CombinedView.swift:232 DBG adding: GPU portal
2024-05-19 17:56:33 CombinedView.swift:232 DBG adding: RAM portal
2024-05-19 17:56:33 CombinedView.swift:232 DBG adding: Disk portal
2024-05-19 17:56:33 CombinedView.swift:232 DBG adding: Network portal

vdhamer, 2025-01-11:
Looks like the M1 Ultra identifies itself like you expected in your code:

[image]

exelban, 2025-01-12:
Yes, please read the comment I share and run the build from that.

vdhamer, 2025-01-12:
> Yes, please read the comment I share and run the build from that.

I first tried 2.11.24 release. No change: still 600 MHz.
Will try special build now.

vdhamer, 2025-01-12:
Special build of 2.11.23.

- has only CPU module (fine)
- generates log.txt (fine)
- timestamps in log file and file date are suspicious

Content:
```
----------------
2025-01-05 07:21:26 readers.swift:316 DBG e-cores freq: [1, 1, 1, 2, 2, 2, 2]
2025-01-05 07:21:26 readers.swift:317 DBG p-cores freq: [1, 1, 1, 2, 2, 2, 2, 3, 3, 3, 3, 3, 3, 3, 3, 4, 4, 4, 4]
2025-01-05 07:21:26 readers.swift:324 DBG ------------- read -------------
2025-01-05 07:21:26 readers.swift:329 DBG min e core freq 1.0, min p core freq 1.0
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.6260706810593568
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 2.788837744533947
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 3.1192513311384062
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.6413256765664048
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.6814567309751238
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.712043762715452
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.7907643372759652
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 3.328087902310772
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 3.2345515069941833
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 3.225874264604141
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 3.365073664679155
2025-01-05 07:21:27 readers.swift:356 DBG eCores: 5 pCores: 0 | e avg freq 1.6903322377184604 p avg freq 1.9061676414260602
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.73536583091455
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.4823328364516144
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.7481419113844523
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.7263346758938902
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.7990290878394593
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.6822429906542056
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.3807094829584976
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.1210401101882383
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.5847219742152066
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.5674949048176574
2025-01-05 07:21:27 readers.swift:356 DBG eCores: 5 pCores: 1 | e avg freq 1.7382228993373114 p avg freq 0.8136299308631214
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.2392138360160303
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.7900085255215739
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.215743777382147
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.1955771133551103
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.2637543969740757
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.159377740496435
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.818927435007606
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.8175976264142557
2025-01-05 07:21:27 readers.swift:356 DBG eCores: 5 pCores: 2 | e avg freq 1.2147333728447596 p avg freq 0.8426533586943435
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.2555302565929047
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.161250227202209
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.2801286895733268
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.2444440898341382
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.3188600275246065
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.248542550701053
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
[... 520 lines cut ...]
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 1.5577208498835153
2025-01-05 07:21:35 readers.swift:356 DBG eCores: 5 pCores: 2 | e avg freq 1.0754108136858658 p avg freq 0.3115441699767031
2025-01-05 07:21:35 readers.swift:343 DBG ECPU 1.1431176383177184
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:35 readers.swift:343 DBG ECPU 1.1158042563769415
2025-01-05 07:21:35 readers.swift:343 DBG ECPU 1.1001045216480776
2025-01-05 07:21:35 readers.swift:343 DBG ECPU 1.2891619928916198
2025-01-05 07:21:35 readers.swift:343 DBG ECPU 1.1794049700973726
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:35 readers.swift:356 DBG eCores: 5 pCores: 3 | e avg freq 1.1655186758663458 p avg freq 0.5
2025-01-05 07:21:35 readers.swift:365 DBG e freq: 1.1198279960350124 p freq: 1.290251928467588
```

litori, 2025-01-12:
M4 still shows 0 and 1 frequency.  How do I disable this feature?

exelban, 2025-01-12:
hmm, it looks similar to m4. Try to run the latest release

exelban, 2025-01-12:
please try this build: 
[Stats.dmg.zip](https://github.com/user-attachments/files/18391067/Stats.dmg.zip)

vdhamer, 2025-01-13:
> please try this build: [Stats.dmg.zip](https://github.com/user-attachments/files/18391067/Stats.dmg.zip)

600 MHz now became 600000 MHz. Hmmm.

[image]

```
----------------
2025-01-05 07:21:26 readers.swift:316 DBG e-cores freq: [1, 1, 1, 2, 2, 2, 2]
2025-01-05 07:21:26 readers.swift:317 DBG p-cores freq: [1, 1, 1, 2, 2, 2, 2, 3, 3, 3, 3, 3, 3, 3, 3, 4, 4, 4, 4]
2025-01-05 07:21:26 readers.swift:324 DBG ------------- read -------------
2025-01-05 07:21:26 readers.swift:329 DBG min e core freq 1.0, min p core freq 1.0
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.6260706810593568
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 2.788837744533947
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 3.1192513311384062
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.6413256765664048
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.6814567309751238
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.712043762715452
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.7907643372759652
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 3.328087902310772
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 3.2345515069941833
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 3.225874264604141
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 3.365073664679155
2025-01-05 07:21:27 readers.swift:356 DBG eCores: 5 pCores: 0 | e avg freq 1.6903322377184604 p avg freq 1.9061676414260602
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.73536583091455
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.4823328364516144
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.7481419113844523
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.7263346758938902
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.7990290878394593
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.6822429906542056
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.3807094829584976
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.1210401101882383
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.5847219742152066
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.5674949048176574
2025-01-05 07:21:27 readers.swift:356 DBG eCores: 5 pCores: 1 | e avg freq 1.7382228993373114 p avg freq 0.8136299308631214
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.2392138360160303
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.7900085255215739
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.215743777382147
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.1955771133551103
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.2637543969740757
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.159377740496435
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.818927435007606
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.8175976264142557
2025-01-05 07:21:27 readers.swift:356 DBG eCores: 5 pCores: 2 | e avg freq 1.2147333728447596 p avg freq 0.8426533586943435
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.2555302565929047
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 1.161250227202209
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.2801286895733268
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.2444440898341382
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.3188600275246065
2025-01-05 07:21:27 readers.swift:343 DBG ECPU 1.248542550701053
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:27 readers.swift:348 DBG PCPU 0.0
[... 519 lines cut ...]
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 1.5577208498835153
2025-01-05 07:21:35 readers.swift:356 DBG eCores: 5 pCores: 2 | e avg freq 1.0754108136858658 p avg freq 0.3115441699767031
2025-01-05 07:21:35 readers.swift:343 DBG ECPU 1.1431176383177184
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:35 readers.swift:343 DBG ECPU 1.1158042563769415
2025-01-05 07:21:35 readers.swift:343 DBG ECPU 1.1001045216480776
2025-01-05 07:21:35 readers.swift:343 DBG ECPU 1.2891619928916198
2025-01-05 07:21:35 readers.swift:343 DBG ECPU 1.1794049700973726
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 0.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:35 readers.swift:348 DBG PCPU 1.0
2025-01-05 07:21:35 readers.swift:356 DBG eCores: 5 pCores: 3 | e avg freq 1.1655186758663458 p avg freq 0.5
2025-01-05 07:21:35 readers.swift:365 DBG e freq: 1.1198279960350124 p freq: 1.290251928467588
```

exelban, 2025-01-13:
ok, so please run command: `ioreg -r -f -x -c AppleARMIODevice`. And find for two values: `voltage-states1-sram` and `voltage-states5-sram`. Please copy just these values and share it here

exelban, 2025-01-13:
it will looks something like that: 
```
"voltage-states1-sram" =
  |     00000000: 00 46 C3 23 35 02 00 00 00 8B EF 39 35 02 00 .F.#5......95..
  |     0000000F: 00 00 B5 64 4F 54 02 00 00 00 FA 90 65 83 02 ...dOT......e..
  |     0000001E: 00 00 00 24 06 7B CE 02 00 00                ...$.{....
```

vdhamer, 2025-01-13:
```
  |   "voltage-states1-sram" = 
  |     00000000: 00 46 C3 23 38 02 00 00 00 8B EF 39 38 02 00 00 00 B5 64 4F 54 02 00 00 00 FA 90 65 86 02 00 00 .F.#8......98.....dOT......e....
  |     00000020: 00 24 06 7B D1 02 00 00 
```

``` 
  |   "voltage-states5-sram" = 
  |     00000000: 00 46 C3 23 00 03 00 00 00 47 5A 31 00 03 00 00 00 48 F1 3E 10 03 00 00 00 64 3F 4D 2C 03 00 00 .F.#.....GZ1.....H.>.....d?M,...
  |     00000020: 00 65 D6 5A 32 03 00 00 00 66 6D 68 4B 03 00 00 00 67 04 76 64 03 00 00 00 68 9B 83 90 03 00 00 .e.Z2....fmhK....g.vd....h......
  |     00000040: 00 84 E9 91 C5 03 00 00 00 85 80 9F 01 04 00 00 00 86 17 AD 2C 04 00 00 00 AF F5 B4 2C 04 00 00 ....................,.......,...
  |     00000060: 00 87 AE BA 2C 04 00 00 00 D8 D3 BC 2C 04 00 00 00 5F 67 C0 2C 04 00 00                         ....,.......,...._g.,...
```

exelban, 2025-01-13:
thanks, I will try to adjust code to these values

vdhamer, 2025-01-13:
Out of curiosity, how can I trust that your software is not writing to such registers (they look like settings relevant for overclockers)?
That could lead to instabilities that are almost impossible to find.
Your software doesn't go through the App Store. And even if I could analyze all your source code, I have no way to know what is in your image files because I or GitHub didn't generate these special builds...

No direct reason to be suspicious. But somebody should ask these questions...

exelban, 2025-01-13:
what do you mean by `writing to such registers`?

fiwswe, 2025-01-13:
@vdhamer The "reg" in "io**reg**" stands for registry not register. See `% man ioreg`: _ioreg – show I/O Kit registry_

I don't think any user level code could directly change CPU frequencies or voltages, etc. If it could, that would constitute a bug in macOS. Overclocking is not something that is done on Macs, especially not on Apple Silicon. And given the tight integration of hardware and macOS as well as the security of macOS it would be exceedingly difficult to achieve.

exelban, 2025-01-13:
oh, there is no writings to any registry, I'm not even sure macOS allows it. Also I would like to know apple silicon could be overclocked)

vdhamer, 2025-01-13:
I don't overclock anything, and recognize that overclocking is usually for serious gaming. Which is so far hardly popular on MacOS.

But I suspect that the values you are looking at are a) written by the CPU into hardware registers, b) because it makes the design more tunable for Apple in late stages of development. The fact that `ioreg` is read-only just means you need another path to get write access. For a discussion on M series software configurability: https://forums.anandtech.com/threads/is-it-possible-to-overclock-an-apple-m1-chip.2600513/

Kapersky writes "On a Mac, keep up to date with new releases. Mac updates don't just add new features – they also remove malware, including rootkits. Apple has built-in security features to protect from malware. However, there are no known rootkit detectors on macOS, so if you suspect a rootkit on your device, you should reinstall macOS. Doing so removes most apps and rootkits on your machine. As above, if the rootkit has infected the BIOS, it will require a repair to fix – and if the rootkit remains, you may need to buy a new device."

My point is that if you adjust anything that isn't really meant to be adjusted by end users, you are at least taking a risk. Arguably even a fan speed reduction, unless a too low fan speed gets compensated by CPU frequency or voltage reduction. My point is that all this starts to be scary stuff as soon as you reconfigure anything. So the question is how safe it is for end users. Are you only using public and thus approved APIs? Any (even temporary) changes to settings other than fan speed?

exelban, 2025-01-13:
Again as @fiwswe it's not about writing, but about reading.
The values I ask you to share it's a list of working frequencies of the cores. It needs to determine in what frequency state core working in the moment. It's not registry, it's just a system report.

Im just curious why you want to overclock Mac, and where you hear that if popular on macOS (cause I never heard about that at all)?

exelban, 2025-01-13:
Basically if you aware about the system I can only recommend to not use any app outside the AppStore at all.
Stats is a monitoring tool that only reads from the system, and not changes in the system. Only exception its a fan control, which I was basically not ok with. But too much people want this feature.

PS: Stats is not published in the AppStore because:
- the name was used
- Stats does not fit the Apple UX requirements, I have some version even fit to the AppStore, but there was too much changes to support it

vdhamer, 2025-01-14:
For what it is worth, I agree with your remarks: I would stress to users that it is (with one exception) a read-only tool.

But it would be good for users and for visibility of the app to have an AppStore version. While keeping the source code on GitHub: good for getting help, good for trustworthiness. The two disadvantages you mention are both opportunities.

Name: Stats is nice and short, but is very ambiguous. The tool is not really associated with Statistics at all: you just collect and show data. In your case you use the term statistics to mean measurements or data. I would something like "system" and "info".

User interface: the use of the title bar is excellent. But Settings behaves very unMac like: you don't use the menu, so the menu shows menus for a wrong tool when Stats Settings has focus.

What you could aim at: charge a small one-time price via the AppStore (it costs you USD 100 yearly just to be able to distribute via the App Store). So charge $1 to $3? Anyone who wants to avoid that, can still use your .dmg image files. Most will see this as "a risk" and "an inconvenience". Benefits: you reach more people, it looks safer to non-experts, it looks safer to IT people, you get some more income, you might get highlighted by the AppStore if you market it well, you get say monthly updates instead of weekly updates (monthly is already very often).
Drawback: you need UI changes to make it Mac like, and you may be asked to remove the fan control feature from the App Store version (if it violates a rule).

Alternatively: create a basic (low cost, simple) version which is enough for 95% of the users. And a Pro version with all the settings you provide for nerds. This avoids trying to be 2 different solutions for 2 audiences in the same app.

Just trying to help: I have no personal benefit if you make these changes. I don't use fan control. I don't use the advanced features. My own (unrelated, mainly iOS) app is both App Store and Github and is entirely free.

exelban, 2025-01-14:
there is one problem: I need to spend like 3-6 months to make app fit AppStore regulations. And still a lot of features will be not available, so it's not worth for me at all.

Also I'm not interested in AppStore, I have literally one app installed from that: Xcode.

exelban, 2025-01-14:
update: I have take a look and I have even accepted build in App Store (build from 2020). But for some reason not published it and drop idea with App Store. 

Also, every build is signed with app dev account. So it's not like I need to buy a developer account, I already have it and use it to sign every build.

I was thinking today about that, and maybe I will make another approach with App Store.

vdhamer, 2025-01-15:
I checked the naming “Stats” for 5 minutes. There are more apps that use “stats” to mean “we give you easy access to info”. So you are not alone there, I tend to stick to the classical meaning: “when there are many instances of X (e.g. patients, cars, user sessions) any aggregation of data across all these instances (e.g. life expectancy of patients, duration of user sessions).

Your 1 line summary in GitHub (“macOS system monitor in your menu bar“) explains well what the app does. I would edit it to “ macOS system monitor on the menu bar”. But that is just a detail. Important to have the app name, slogan and icon well aligned. So the name should echo what is in the slogan. If you change the icon, users may think it is a different app, so best to keep the icon, or to change the icon a year after a name change.

I like that you are thinking with an open mind about putting your app on the App Store. Say you sell 1000 @ 1 $. It might be worth more, just an example. Then Apple early 1000 $ - 30% for you. I expect you don’t get that kind of number via Patreon. And if the app ever starts showing up in reviews or being featured, the numbers might be 10x higher (there are not too many macOS users, and as you mentioned, the App Store there must be very quiet compared to iOS).

moebis, 2025-01-17:
I've seeing 600mhz on my M1 Ultra too, was there a new build that fixed this?

[image]

vdhamer, 2025-01-17:
Not that I know of. But I provided some requested logging data, so I think it will be fixed in a release or two: the issue has a label “needs fixing” ;-)

image.png (view on web)<https://github.com/user-attachments/assets/1d7884ef-e920-4ea7-a8c2-ca9addace69d>

exelban, 2025-01-17:
@moebis but maybe you can help. Please take a look at the comment I mentioned before and run that build and share the logs. 

Cause I'm looking right now at the logs @vdhamer shared and it's for 100% not his logs. So I spend time looking at CPU states without any reason....

vdhamer, 2025-01-17:
> @vdhamer<https://github.com/vdhamer> I'm just curious right now, why you shared the log from m4 pro that was shared by another user in another issue?

Certainly not my intention. Don’t know what happened. Maybe compared both and sent you the wrong one. So must be an error. Want me to generate it again, and right this time?

I don’t have an M4 Mac. I only own an M1 Ultra Mac Studio (don’t have a notebook).

vdhamer, 2025-01-18:
> Certainly not my intention. Don’t know what happened. Maybe compared both and sent you the wrong one. So must be an error. Want me to generate it again, and right this time?

Found it the file you were expecting. I had sent you something from Downloads instead of Documents folder.
That explains why the date looked suspicious. Sorry for error.

----------------
2024-05-15 10:09:26 CombinedView.swift:229 DBG modules: 5
2024-05-15 10:09:26 CombinedView.swift:232 DBG adding: CPU portal
2024-05-15 10:09:26 CombinedView.swift:232 DBG adding: GPU portal
2024-05-15 10:09:26 CombinedView.swift:232 DBG adding: RAM portal
2024-05-15 10:09:26 CombinedView.swift:232 DBG adding: Disk portal
2024-05-15 10:09:26 CombinedView.swift:232 DBG adding: Network portal
----------------
2024-05-15 10:25:45 CombinedView.swift:229 DBG modules: 5
2024-05-15 10:25:45 CombinedView.swift:232 DBG adding: CPU portal
2024-05-15 10:25:45 CombinedView.swift:232 DBG adding: GPU portal
2024-05-15 10:25:45 CombinedView.swift:232 DBG adding: RAM portal
2024-05-15 10:25:45 CombinedView.swift:232 DBG adding: Disk portal
2024-05-15 10:25:45 CombinedView.swift:232 DBG adding: Network portal
----------------
2024-05-17 19:09:39 CombinedView.swift:229 DBG modules: 5
2024-05-17 19:09:39 CombinedView.swift:232 DBG adding: CPU portal
2024-05-17 19:09:39 CombinedView.swift:232 DBG adding: GPU portal
2024-05-17 19:09:39 CombinedView.swift:232 DBG adding: RAM portal
2024-05-17 19:09:39 CombinedView.swift:232 DBG adding: Disk portal
2024-05-17 19:09:39 CombinedView.swift:232 DBG adding: Network portal
----------------
2024-05-18 14:33:57 CombinedView.swift:229 DBG modules: 5
2024-05-18 14:33:57 CombinedView.swift:232 DBG adding: CPU portal
2024-05-18 14:33:57 CombinedView.swift:232 DBG adding: GPU portal
2024-05-18 14:33:57 CombinedView.swift:232 DBG adding: RAM portal
2024-05-18 14:33:57 CombinedView.swift:232 DBG adding: Disk portal
2024-05-18 14:33:57 CombinedView.swift:232 DBG adding: Network portal
----------------
2024-05-19 10:16:02 CombinedView.swift:229 DBG modules: 5
2024-05-19 10:16:02 CombinedView.swift:232 DBG adding: CPU portal
2024-05-19 10:16:02 CombinedView.swift:232 DBG adding: GPU portal
2024-05-19 10:16:02 CombinedView.swift:232 DBG adding: RAM portal
2024-05-19 10:16:02 CombinedView.swift:232 DBG adding: Disk portal
2024-05-19 10:16:02 CombinedView.swift:232 DBG adding: Network portal
----------------
2024-05-19 17:56:33 CombinedView.swift:229 DBG modules: 5
2024-05-19 17:56:33 CombinedView.swift:232 DBG adding: CPU portal
2024-05-19 17:56:33 CombinedView.swift:232 DBG adding: GPU portal
2024-05-19 17:56:33 CombinedView.swift:232 DBG adding: RAM portal
2024-05-19 17:56:33 CombinedView.swift:232 DBG adding: Disk portal
2024-05-19 17:56:33 CombinedView.swift:232 DBG adding: Network portal

If needed, I can reload the special patched version and send a fresher file.

moebismaccom, 2025-01-27:
@exelban yep this is still happening with the last 2 releases

moebismaccom, 2025-01-28:
@exelban still happening in 2.11.28 .... iStats Menu doesn't have this issue, it properly reports the current CPU freq. are there any logs or diagnostics I can provide?

exelban, 2025-01-29:
Hi. To start I need a logs from the build of this post https://github.com/exelban/stats/issues/2285#issuecomment-2571324324

vdhamer, 2025-02-08:
Log for my M1 Ultra which keeps reporting 600 MHz on new versions of Stats.
When you create test releases, please label these versions clearly.

```
----------------
2025-02-08 23:23:11 readers.swift:316 DBG e-cores freq: [600, 972, 1332, 1704, 2064]
2025-02-08 23:23:11 readers.swift:317 DBG p-cores freq: [600, 828, 1056, 1296, 1524, 1752, 1980, 2208, 2448, 2676, 2904, 3036, 3132, 3168, 3228]
2025-02-08 23:23:11 readers.swift:324 DBG ------------- read -------------
2025-02-08 23:23:11 readers.swift:329 DBG min e core freq 600.0, min p core freq 600.0
2025-02-08 23:23:12 readers.swift:356 DBG eCores: 0 pCores: 0 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:12 readers.swift:356 DBG eCores: 0 pCores: 1 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:12 readers.swift:356 DBG eCores: 0 pCores: 2 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:12 readers.swift:356 DBG eCores: 0 pCores: 3 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:12 readers.swift:365 DBG e freq: 600.0 p freq: 600.0
2025-02-08 23:23:12 readers.swift:324 DBG ------------- read -------------
2025-02-08 23:23:12 readers.swift:329 DBG min e core freq 600.0, min p core freq 600.0
2025-02-08 23:23:13 readers.swift:356 DBG eCores: 0 pCores: 0 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:13 readers.swift:356 DBG eCores: 0 pCores: 1 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:13 readers.swift:356 DBG eCores: 0 pCores: 2 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:13 readers.swift:356 DBG eCores: 0 pCores: 3 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:13 readers.swift:365 DBG e freq: 600.0 p freq: 600.0
2025-02-08 23:23:13 readers.swift:324 DBG ------------- read -------------
2025-02-08 23:23:13 readers.swift:329 DBG min e core freq 600.0, min p core freq 600.0
2025-02-08 23:23:14 readers.swift:356 DBG eCores: 0 pCores: 0 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:14 readers.swift:356 DBG eCores: 0 pCores: 1 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:14 readers.swift:356 DBG eCores: 0 pCores: 2 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:14 readers.swift:356 DBG eCores: 0 pCores: 3 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:14 readers.swift:365 DBG e freq: 600.0 p freq: 600.0
2025-02-08 23:23:14 readers.swift:324 DBG ------------- read -------------
2025-02-08 23:23:14 readers.swift:329 DBG min e core freq 600.0, min p core freq 600.0
2025-02-08 23:23:15 readers.swift:356 DBG eCores: 0 pCores: 0 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:15 readers.swift:356 DBG eCores: 0 pCores: 1 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:15 readers.swift:356 DBG eCores: 0 pCores: 2 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:15 readers.swift:356 DBG eCores: 0 pCores: 3 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:15 readers.swift:365 DBG e freq: 600.0 p freq: 600.0
2025-02-08 23:23:15 readers.swift:324 DBG ------------- read -------------
2025-02-08 23:23:15 readers.swift:329 DBG min e core freq 600.0, min p core freq 600.0
2025-02-08 23:23:16 readers.swift:356 DBG eCores: 0 pCores: 0 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:16 readers.swift:356 DBG eCores: 0 pCores: 1 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:16 readers.swift:356 DBG eCores: 0 pCores: 2 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:16 readers.swift:356 DBG eCores: 0 pCores: 3 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:16 readers.swift:365 DBG e freq: 600.0 p freq: 600.0
2025-02-08 23:23:16 readers.swift:324 DBG ------------- read -------------
2025-02-08 23:23:16 readers.swift:329 DBG min e core freq 600.0, min p core freq 600.0
2025-02-08 23:23:17 readers.swift:356 DBG eCores: 0 pCores: 0 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:17 readers.swift:356 DBG eCores: 0 pCores: 1 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:17 readers.swift:356 DBG eCores: 0 pCores: 2 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:17 readers.swift:356 DBG eCores: 0 pCores: 3 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:17 readers.swift:365 DBG e freq: 600.0 p freq: 600.0
2025-02-08 23:23:17 readers.swift:324 DBG ------------- read -------------
2025-02-08 23:23:17 readers.swift:329 DBG min e core freq 600.0, min p core freq 600.0
2025-02-08 23:23:18 readers.swift:356 DBG eCores: 0 pCores: 0 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:18 readers.swift:356 DBG eCores: 0 pCores: 1 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:18 readers.swift:356 DBG eCores: 0 pCores: 2 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:18 readers.swift:356 DBG eCores: 0 pCores: 3 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:18 readers.swift:365 DBG e freq: 600.0 p freq: 600.0
2025-02-08 23:23:18 readers.swift:324 DBG ------------- read -------------
2025-02-08 23:23:18 readers.swift:329 DBG min e core freq 600.0, min p core freq 600.0
2025-02-08 23:23:19 readers.swift:356 DBG eCores: 0 pCores: 0 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:19 readers.swift:356 DBG eCores: 0 pCores: 1 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:19 readers.swift:356 DBG eCores: 0 pCores: 2 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:19 readers.swift:356 DBG eCores: 0 pCores: 3 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:19 readers.swift:365 DBG e freq: 600.0 p freq: 600.0
2025-02-08 23:23:19 readers.swift:324 DBG ------------- read -------------
2025-02-08 23:23:19 readers.swift:329 DBG min e core freq 600.0, min p core freq 600.0
[... cut ...]
2025-02-08 23:23:32 readers.swift:356 DBG eCores: 0 pCores: 1 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:32 readers.swift:356 DBG eCores: 0 pCores: 2 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:32 readers.swift:356 DBG eCores: 0 pCores: 3 | e avg freq 0.0 p avg freq 0.0
2025-02-08 23:23:32 readers.swift:365 DBG e freq: 600.0 p freq: 600.0
2025-02-08 23:23:32 readers.swift:324 DBG ------------- read -------------
2025-02-08 23:23:32 readers.swift:329 DBG min e core freq 600.0, min p core freq 600.0

```

vdhamer, 2025-02-09:
Just in case it is useful, log on a friend's MacBook Pro with M4 chip.
Stats shows realistic-looking frequencies there.

```
----------------
2025-02-09 22:56:13 readers.swift:316 DBG e-cores freq: [0, 1, 1, 1, 2, 2, 2, 2]
2025-02-09 22:56:13 readers.swift:317 DBG p-cores freq: [0, 1, 1, 1, 2, 2, 2, 2, 3, 3, 3, 3, 3, 3, 3, 3, 4, 4, 4, 4]
2025-02-09 22:56:13 readers.swift:324 DBG ------------- read -------------
2025-02-09 22:56:13 readers.swift:329 DBG min e core freq 0.0, min p core freq 0.0
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.5740050239718681
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 2.874264180786959
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.6059119621346365
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.6173120652343238
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.6474323507811692
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.7397493275251001
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.8018707005418904
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.7176152451506794
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 3.024881922312632
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 3.322451798081585
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 3.649622822127173
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 3.4873012148620512
2025-02-09 22:56:14 readers.swift:356 DBG eCores: 7 pCores: 0 | e avg freq 1.6719852393342383 p avg freq 3.27170438763408
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.2442122888834508
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 1.5838713078412767
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.2572541103058796
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.2651697301285532
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.2906268585893077
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.3586045159950308
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.3568847838539515
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.3862019067649034
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 0.0
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 1.594495359134163
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 0.0
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 1.6409344188066568
2025-02-09 22:56:14 readers.swift:356 DBG eCores: 7 pCores: 1 | e avg freq 1.308422027788725 p avg freq 0.9638602171564192
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.6140800287084442
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 3.4914139407583984
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.6254994599635253
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.6040850926280814
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.618937678637973
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.6504810636060758
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.711218273946054
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.6837084802729574
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 4.0
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 3.4409069842062108
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 0.0
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 3.5723665097902506
2025-02-09 22:56:14 readers.swift:356 DBG eCores: 7 pCores: 2 | e avg freq 1.6440014396804443 p avg freq 2.900937486950972
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.262599026150556
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 0.0
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.272376392117424
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.2101992913580792
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.2660923378764792
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.1873820616568591
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.2850326567327623
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.134222774058594
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 0.0
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 0.0
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 0.0
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 0.0
2025-02-09 22:56:14 readers.swift:356 DBG eCores: 7 pCores: 3 | e avg freq 1.2311292199929649 p avg freq 0.0
2025-02-09 22:56:14 readers.swift:365 DBG e freq: 1.4638844816990932 p freq: 1.7841255229353679
2025-02-09 22:56:14 readers.swift:324 DBG ------------- read -------------
2025-02-09 22:56:14 readers.swift:329 DBG min e core freq 0.0, min p core freq 0.0
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.367313425964533
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 1.760236007724983
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.379728083925992
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.3760351220843263
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.365586112023425
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.3915989152270893
[... 23 lines cut ...]
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.1314058174371422
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.1566680463291312
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.202225883569167
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.2492302814110186
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.1577966881324748
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 1.0
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 0.9773078572758098
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 1.0
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 1.0
2025-02-09 22:56:14 readers.swift:356 DBG eCores: 7 pCores: 2 | e avg freq 1.1828284579078594 p avg freq 0.990923142910324
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.0
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 0.0
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.0
[... cut ...]
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.0
2025-02-09 22:56:14 readers.swift:343 DBG ECPU 1.0
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 0.0
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 0.0
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 0.0
2025-02-09 22:56:14 readers.swift:348 DBG PCPU 0.0
2025-02-09 22:56:14 readers.swift:356 DBG eCores: 7 pCores: 3 | e avg freq 1.0 p avg freq 0.0
2025-02-09 22:56:14 readers.swift:365 DBG e freq: 1.2244061223631593 p freq: 0.9635114692675153
```

closed 2025-06-23 by exelban