exelban/stats #2331: Fan Helper (not Stats) cannot be uninstalled

jhmaster2000, 2025-01-13:
**Describe the bug**
[!] PLEASE READ, I HAVE READ THE FAQ AND AM AWARE OF FAN CONTROL BEING DEPRECATED:
This report concerns either a bug or confusing implementation of how the deprecation is being applied for existing Stats installations, it is not in any way requesting any support with *using* the fan control feature. I am explicitly bringing this up first and foremost because whilst searching if this had already been reported, it has, in a few different ways: #2235, #1980 and #1650, but all seem to have been misinterpreted and blanket-closed for *appearing* to be related to wanting to use fan control, *that is not the case, it is the exact opposite!*.

Stats is an amazing app, and I do NOT wish to uninstall it! I want to KEEP the Stats app itself, what I am attempting to do is uninstall the **Fan Helper** program ONLY, given I have read the feature is deprecated therefore I should have no need for the Fan Helper program anymore then. Unfortunately, this is where the actual bug comes in:

Even though Stats has an option in the Settings menu called "Uninstall Fan Helper", as described in the 3 previous issues linked above, that button simply instantly disappears when clicked, giving no confirmation or sign of having done anything, and upon fully closing Stats (Stats menu -> Power button) and re-opening it, the "Uninstall Fan Helper" button re-appears, implying it is still installed and was not uninstalled.

It is unclear whether Fan Helper is actually not being uninstalled, or if it was successfully uninstalled and Stats is simply failing to register that it was uninstalled and just erroneously thinks its still installed, I do not know where Stats places its logs or where Fan Helper is installed at, but if given this info I'll gladly check to provide further details on what actually is the case.

**Details:**
 - Device: Macbook Pro 16" M3 Max (14 core)
 - macOS: Sequoia 15.2
 - Application version: 2.11.24

jhmaster2000, 2025-01-13:
**Update:** I do not know Swift, but doing a simple search for "fan helper" strings around the codebase I was able to find this function: https://github.com/exelban/stats/blob/307465ad3078e59316f64cd3e10e985c4181f078/Kit/helpers.swift#L863
so I've ran the `ls` command myself in Terminal and indeed, even though I have pressed "Uninstall Fan Helper" in Stats, the `eu.exelban.Stats.SMC.Helper` is still present in there.

exelban, 2025-01-13:
so, the fan helpers is uninstalled or not?
I'm loosed a little bit

jhmaster2000, 2025-01-13:
I was able to *manually*, outside of Stats, get Fan Helper to uninstall itself by finding where it was installed, so I've personally solved my own problem with that *workaround*, but Stats still has the relevant bugs, which are:
1. Stats fails to uninstall Fan Helper from within the app with the "Uninstall Fan Helper" button, the button doesn't do anything.
2. Stats fails to detect Fan Helper is uninstalled after manual uninstall. (It still shows the uninstall button)

jhmaster2000, 2025-01-13:
Is there somewhere Stats writes logs to that I could provide a log of what might be happening under the hood?

Step by step with additional full context of my journey here:
1. I first installed Stats 2.9.x on MacOS Sonoma (14)
2. I disabled auto-update checking of Stats
3. I installed Fan Helper through Stats and it used to work well back then
4. I update my Mac to Sequoia (15) and then to 15.1 and 15.2 as they came
6. One day notice fan control doesn't work anymore (because of MacOS 15 presumably), but I wasn't really using it anyway so I just ignored it and moved on (and I'd forgotten it used a helper)
7. Earlier this year I decide to re-arrange a few of my Stats modules, and decide to check for updates, and upgrade to Stats 2.11.24 (skipping 2.10.x, if that may be of relevance?) through Stats' self-updater
8. I came here to check out the changelogs I missed from 2.9 -> 2.11, and spotted the one stating fan control is deprecated.
9. In light of that, and remembering it used a helper, I decide to uninstall the helper given its no longer needed, so I find the "Uninstall Fan Helper" button and try to use it
10. The button disappears seemingly doing nothing.
11. Restart Stats, "Uninstall Fan Helper" button re-appears
12. And that brings us to this issue

NostromoUS, 2025-01-19:
I too have this issue. I installed Stats just to monitor temps and fan speed. I do not with to control fans or change any of the default system behavior when it comes to fan curve etc. I basically want to get rid of Fan Helper but keep Stats of course. 

If it's deprecated maybe it should be included as an optional install and not forced?

closed 2025-01-26 by exelban

jhmaster2000, 2025-03-19:
Even though this was closed as "not planned", I just wanted to update here that at some point between 2.11.24 and 2.11.34 this seems fixed for me.

I don't have automatic updates enabled so updated all of those versions at once so I can't know which one exactly it might have been, but after the update I tried one more time to click the button to uninstall the helper and it finally disappeared, so that's nice "accidental" fix I suppose.