exelban/stats #2347: Include a weather module

raykroeker, 2025-01-21:
I'd love to see a weather module included.  I'm not sure if this has already been requested, but a single representation of the local weather or multiple of static locations (similar to time).

Thanks!

exelban, 2025-01-26:
It was, but I was not ready to add this feature. But for some time I missing that too, so I suppose it will be added in near feature. The biggest problem I have with that is there is no good free API for weather.

exelban, 2025-02-07:
@PuzzledUser wow, I don't even know. So yes, there is no any reason to add that to the app.

closed 2025-02-07 by exelban

PuzzledUser, 2025-02-07:
Wonderful, perhaps with the time you save you might consider: https://github.com/exelban/stats/issues/1933 😉