exelban/stats #2348: App no longer opening after changing settings

AhmadHrb, 2025-01-22:
**Describe the bug**
Hello, so the app have been working for months but today I was changing in the settings and I changed disk usage "Character" to "Arrows" and the app crashed and no longer works no matter what I do, I think it's related to showing only "Write" Value and with arrows. I tried to delete ~/Library/Application Support/Stats but that didn't help.

**Details:**
 - Device: MacBook Air M3
 - macOS: 15.2
 - Application version: latest version (I cant check atm)

exelban, 2025-01-22:
You can reset the settings with `defaults delete eu.exelban.Stats`. After please check the version, because this kind of bug was fixed some time ago.

AhmadHrb, 2025-01-22:
Im on version **2.11.23** I thought im on latest because usually it prompts me to update but it didn't recently so im not sure.
Thanks it works now if the fix was after this version then it's all fine. Otherwise I can provide the logs as needed

exelban, 2025-01-23:
@AhmadHrb does it fix your case?

closed 2025-01-26 by exelban