exelban/stats #2363: Date and rows mechanic has changed in latest version

PatDal81, 2025-01-26:
See differences between the 2 last versions:

2.11.26:

[image]

2.11.27:

[image]

Here's the Widget options for this tab

[image]

PatDal81, 2025-01-26:
You can ignore the Module "Clock 2" as I used it for testing purposes.

GiorgioMendieta, 2025-01-26:
I also hae an issue where my added clocks are not displaying correctly on the menu bar

Shunsui7, 2025-01-27:
Noticed it as well after the update.. Sucks that downgrade is the only option but I believe it will get sorted.

GiorgioMendieta, 2025-01-27:
Hey, I just updated to 2.11.28 and it is working again, thanks!)

Shunsui7, 2025-01-27:
Sensational!!

exelban, 2025-01-27:
sorry for that, pretty stupid bug(

closed 2025-01-27 by exelban

PatDal81, 2025-01-27:
I confirm the issue is resolved with latest version.  Thanks for the prompt response!