exelban/stats #2372: Smushed Stats

whoisstan, 2025-01-30:
I am seeing this on my 2021 M1 running 15.3. Similar when enabling the separator.
[image]

whoisstan, 2025-01-31:
Now it works, it seemed that the inital setup wizard started much after I opened the app.

exelban, 2025-01-31:
hi. Wow, never seen the bug like this. Could you give more details?
Or maybe could you share your settings of the Stats? It could be exported in the Stats settings.

isobelbutler, 2025-01-31:
I just had the exact same issue as this on my M3 2024 running 15.1. Occurred when I downloaded the app via the following commands:
`
brew install stats
open /Applications/Stats.app
`

The app opened (could see smushed icons in menubar when toggling 'Combined modules' in settings, otherwise no icons) but it wasn't registering any stats (all 0% when clicking smushed icon/no tabs for the stats showing in the app itself).

Fixed the issue it by selecting the 'Reset settings' option in settings which ran me through the setup (I wasn't initially taken through setup).

whoisstan, 2025-01-31:
Yes! I wasn’t initially taken through the setup as well, eventually it did with the reset. Sent from my iPhone