exelban/stats #2375: Incorrect country flag displayed (GB) for VPN IP in Switzerland

scottnzuk, 2025-02-01:
**Description:**
I have encountered a bug where the stats software displays my public IP address alongside the country flag (GB). However, this IP is actually provided by my VPN, which is based in Switzerland. I am wondering where the country information is being sourced from, as it appears to be inaccurate.

**Steps to Reproduce:**
	1.	Connect to a VPN service with an exit node in Switzerland.
	2.	Launch the stats software.
	3.	Observe that your public IP is shown with the country code (GB) instead of (CH) or the correct flag for Switzerland.
	4.	Confirm using an external IP lookup that the VPN IP is indeed from Switzerland.

**Expected Behaviour:**
The software should display the correct country flag/code (e.g. (CH) for Switzerland) corresponding to the actual location of the IP.

**Actual Behaviour:**
The stats software shows the public IP with the (GB) flag, which is misleading given that the VPN is based in Switzerland.

**Additional Information:**
	•	It’s unclear where the country information is being retrieved from – whether it’s via a third-party API or a local database.
	•	Please could you clarify the source of the country data in the code?

[image]

exelban, 2025-02-01:
it's a data from macOS system.

scottnzuk, 2025-02-01:
1.	MaxMind’s GeoLite2 Country Database
This database is free (registration is required) and it comes in CSV format. It includes IPv4 and IPv6 blocks assigned to countries. Note that MaxMind updates the data periodically so you may want to set up a process to update your copy.
	2.	IP2Location LITE Database
This is another free database that maps IP ranges to countries and is available in a couple of formats including CSV.

semi easy fix, use one of free geoip local db files.

it has me worried my VPN was not working correctly. or perhaps remove it if false data?

exelban, 2025-02-01:
thx for sharing these dbs. I will take a look and maybe will change from native solution to this one.