exelban/stats #2377: Can't change colors of menu bar modules?

DarthZiplock, 2025-02-02:
I was using an older version of Stats on macOS Mojave for a long time, and recently upgraded to an M2 Pro Mac Mini on macOS Sequoia.

On the older version, I was able to change the colors of the modules and set my CPU meter so it was white bars on a transparent/black backgroud.

On the current version I no longer see the color-changing option, and it's impossible to see the default green bars on the white box backgroud.

Was this feature removed? If so, please bring it back. It's a huge downgrade in usability.

exelban, 2025-02-03:
hi. No colorization option was removed as I can remember. What exactly widget you mean? And what option you are looking for?
Will be much easier if you share the screenshot what you want to achieve.

DarthZiplock, 2025-02-03:
I edited my OP cuz I'm a moron and wasn't looking in the right spot. Color options are in the "Widgets" tab of their respective module configs, with options to turn off the box and border. That's what I was looking for and was blind apparently.

Everything is working as it should. Don't mind me, sorry for wasting your time.

exelban, 2025-02-03:
np

closed 2025-02-03 by exelban