exelban/stats #2382: Stats crashese everytime when I try to install the offered update

CodingMarkus, 2025-02-04:
**Describe the bug**
Every time there is an update, Stats just crashes when I select to install the update.

When I do a manual update check from within the app, it offers me to download the update but then says it cannot install it because it has no write permission to the app folder, which is correct, but not a problem for other apps. Other apps will prompt me for an admin password and then just install it. The app folder should actually not be writeable for normal users, otherwise malware could just overwrite any app there with a malware version.

But even if that folder is not writeable, clicking on the update notification should not crash the Stats app to begin with.

**Details:**
 - Device: MacBook Pro 2021
 - macOS: 14.6.1
 - Application version: Any

Crash log (actually an IPS file, you need to remove .txt to watch it but I GitHub won't let me upload it with .ips extension):

[Stats-2025-02-04-150706.ips.txt](https://github.com/user-attachments/files/18658286/Stats-2025-02-04-150706.ips.txt)

exelban, 2025-02-04:
If you know the way of how app could be replaced in the Application folder without write permission to this folder I would like to know the method.

CodingMarkus, 2025-02-04:
Most other apps either ship their updates using an installer package (`.pkg`), use the Sparkle framework, or have their own privileged helper.

## Options

### Installer Package

If you have a signed installer package (`.pkg`), you can have the system installing it using

```
installer -pkg "/path/to/your.pkg" -target /
```

Note that an installer does not really need to install the files directly, an installer has a pre-install and post-install script. In between it would copy files but I've seen installer that copy nothing (at least nothing meaningful) or just install files to `/tmp` and the actually installation then happens in the scripts (which copy the files from `/tmp` to the real destination or unarchive the real files to install from an archive copied to `/tmp`). The scripts will run as root and the user is prompted to provide root access for the installer, so an installer can install to directories only root has write access to.

But if the installer package contains a real app, the installer will install it and it will only install it to `/Applications` on first install. If the app already exists, it will ask LaunchServices where the app is found and install it to the location the app is currently located (again, prompting for root permissions if required). It finds the already installed app via the app bundle identifier (all apps known to the system are indexed by their bundle identifier in the LaunchServices DB).

### Sparkle

Everyone has at least once seen the [Sparkle installer](https://github.com/sparkle-project/Sparkle). It takes care of automatically updating apps and it will prompt for permissions as required.

To use Sparkle, your app must link against the Sparkle framework and also ship with that framework. Also you require a HTTP server that provides a Sparkle update stream (a simple RSS feed that contains entries for updates, with version number, update link and release notes). Sparkle will check that feed for you, display the release notes, download the update, and install it and replace your current app copy.

### Privileged Helper Tool

Using [SMJobBless](https://developer.apple.com/documentation/servicemanagement/smjobbless(_:_:_:_:)) or [SMAppService](https://developer.apple.com/documentation/servicemanagement/smappservice) on newer systems, an app can install a priviledged helper tool to `/Library/PrivilegedHelperTools` that can then execute with root permissions and perform whatever job you want it to perform. That's how Sparkle is doing the updates. Some apps have their own helper tool for that job.

A privileged helper tool is just an executable binary with an embedded `launchd` plist (you can tell Xcode to embed plists into binaries when linking them; this way a binary that is not an app bundle can still have a Info.plist as if it was an app bundle but in that case it needs a [launchd plist](https://alvinalexander.com/mac-os-x/launchd-examples-launchd-plist-file-examples-mac/)).

`SMJobBless` copies your tool to `/Library/PrivilegedHelperTools`, installs the embedded plist to `/Libaray/LaunchDeamons`, and tells `launchd` to load that plist. What happens next depends on what's inside the plist. E.g. the tool may launch at once or `launchd` may create an IPC port or socket for you that your app can connect to (in that case it will launch on connect) and that way your app can send data to the helper tool. The helper tool runs with root privileges.

Sparkle works like that: Download the update, install the helper tool, let the system start the helper tool, tell it where the update is found, quit the app. Now the helper tool installs the update and finally restarts the app again, before removing itself again (`SMJobRemove` or maybe it just tells `launchd` to forget about the job and deletes its plist and then itself before quitting, which is basically what `SMJobRemove` actually does).

## I'd be glad to help out

Let me know which option you'd prefer and I'd be glad to help out with some code. 

I know how to create signed installers (I've been doing that in the past for my employer). I'm not a big fan of installers but they are easy to use (not necessarily easy to build but once you have a correct package, integrating them into the app should usually not be a big issue) and they do the job.

I have also integrated Sparkle into projects in the past; but Sparkle is no option, if you have no server that could provide the update RSS feed, as I think that is still required.

And I've written and integrated privileged helper tools in the past. Also I'm quite fit with `launchd` and IPC on macOS. This is certainly the most elegant and most customizable option but probably also the one with the highest workload (though Sparkle can also be time consuming if you need customization and cannot use it out of the box).

There are alternatives to Sparkle; I know that those exists but I have zero experience with those and cannot even tell you the name of one. But you may want to have a look for those yourself first.

exelban, 2025-02-05:
@CodingMarkus have you even look at the code?

CodingMarkus, 2025-02-05:
No, I have not looked at the code.

And I searched for "update" and "crash" before creating this issue but there was no hit, as all the crash related issues were crashes on launch or after an update or crashes out of the blue but not crashes when you hit update.

Again: Stats crashes every time it offers me an update and I choose to install it. And when I try to manually perform the update from the settings, it downloads the update, but then tells me that it cannot write the to Applications folder (which may or may not be the cause of the crash). Both should not happen and both won't happen if you use any of the methods I described above.

You asked me "If you know the way of how app could be replaced in the Application folder without write permission to this folder" and I provided you with three methods that will allow you to do just that.

exelban, 2025-02-05:
you provide 0 methods the can do update without access to the Application folder.

CodingMarkus, 2025-02-05:
> you provide 0 methods the can do update without access to the Application folder.

Again, all three methods can update apps in my application folder. My applications folder is owned by root:admin, as it should be, but my user is neither root nor an admin user. 

Other apps use installers or Sparkle or an own helper tool and in all three cases the system will prompt me for an admin password, just like when I copy something in Finder to Applications, then Finder also prompts me for an admin password.

The installer does so automatically when it needs more privileges. The system does so when you install a privileged helper tool. And Sparkle does so as well, since it also uses a privileged helper tool. 

So all these three methods have no issue to update apps in my application folder. Stats is the only app I have that fails to update itself, as it is the only app that is not using one of those three established methods to update itself.