exelban/stats #2387: Inconsistent CPU performance core temperature

Delphineca, 2025-02-06:
**Describe the bug**
The CPU performance core temperature for all cores reported as 40°C regardless of system load. The app sometimes briefly shows the correct high temperature (around 80°C under load) and then promptly falls back to 40°C

**Details:**
 - Device: MacBook Pro 16" (M4 Max)
 - macOS: macOS Sequoia (15.3.0)
 - Application version: v2.11.29

Ben-Jian, 2025-02-08:
Same issue on M4 pro. Additionally, the temperature for CPU performance core 5 is missing.

exelban, 2025-02-09:
hi. Yes, it's a long time of strange sensors behavior on apple soc. The temperature sensors are kind of activating starting from some level. Nothing could be done here.

closed 2025-02-09 by exelban