exelban/stats #2388: Memory leaks on apple chip device

kussumma, 2025-02-10:
**Describe the bug**
Thanks for the awsome app. Usually Stats use less than 100 mb RAM, but now it use > 400 mb RAM

**Details:**
 - Device: Macbook Air M3 chip
 - macOS: 15.3
 - Application version: 2.11.30 (latest)
 

[image]

[image]

Erd89, 2025-02-10:
Same problem on macBook m2pro with only the network sensor active.

mathieufrh, 2025-02-10:
I am on MacBook Pro M4 and can confirm Stats is using about 500 GB of RAM, which is insane and not usual.

exelban, 2025-02-10:
I find the problem, it's not a memory leak. I have add a stress tests for CPU and GPU to the app. And GPU class allocate a memory at start, and it's just takes a lots of RAM on init.

Will be fixed today/tomorrow.

https://github.com/exelban/stats/blob/706a996fafadbc3b5ddc9cce726249fdba81ac60/Kit/helpers.swift#L1749-L1756

exelban, 2025-02-14:
as usual, Sunday

closed 2025-02-16 by exelban