exelban/stats #2391: CPU Stress testing doesn't work

Mahtabctg, 2025-02-10:
**Describe the bug**
A clear and concise description of what the bug is.

**Details:**
 - Device: Macbook Pro 14 M1 pro (2021)
 - macOS: 15.3.1
 - Application version: 2.11.30

Started CPU stress testing for performance and efficiency cores, none worked. GPU Stress testing is working fine.

[image]

ArnavVarshney, 2025-02-21:
@exelban thanks for working on this. just tried the new version of the app but I'm seeing a similar behaviour.

[image]

[image]

exelban, 2025-02-21:
yeap, looks like compiler too aggressive and optimize stress test code(

Mahtabctg, 2025-02-21:
Still the same, need some other way to make it work if compiler optimizes it every time. 

[image]

closed 2025-03-02 by exelban

ArnavVarshney, 2025-03-02:
can confirm it's working now. thanks @exelban