exelban/stats #2394: Stats App Unauthorized After macOS 15.3.1 Update

scottnzuk, 2025-02-11:
**Description:**
After updating my Mac to macOS 15.3.1, the Stats application is flagged as “unauthorized,” prompting me to move it to the trash. I can manually override this warning, but it is unusual behavior.

This issue only started after I ran brew update and ensured I was on the latest version of Stats (2.11.30). Homebrew confirms that the app is installed correctly, but the system still flags it incorrectly.

**Steps to Reproduce:**
	1.	Update macOS to 15.3.1
	2.	Run brew update and brew upgrade
	3.	Attempt to launch Stats
	4.	macOS displays a security warning

**Expected Behavior:**
Stats should open without macOS flagging it as unauthorized.

**Actual Behavior:**
macOS prompts to move the app to the trash, requiring manual override.

**

[image]

**
	•	macOS Version: 15.3.1
	•	Stats Version: 2.11.30 (installed via Homebrew Cask)
	•	Installation Path: /opt/homebrew/Caskroom/stats/2.11.30

(.venv) byteme@ByteMe ~ % xattr -l /Applications/Stats.app
com.apple.provenance: 
com.apple.quarantine: 0381;67abad07;;5D9EC4B1-B8A4-488F-95F0-5BEF660057B6
(.venv) byteme@ByteMe ~ %

exelban, 2025-02-12:
Every build in the GitHub releases is notarized with Apple. I have no response and relation to the build via homebrew.

scottnzuk, 2025-02-12:
Sorry if that sounds stupid but are you saying that potentially the security risk identified in the Doc version could be legitimate and potentially could be spyware and that I shouldn't be using the brew version but direct from website? is brew version not just a scripted install of your own packages?

checktext00, 2025-02-21:
> Sorry if that sounds stupid but are you saying that potentially the security risk identified in the Brew version could be legitimate and potentially could be spyware and that I shouldn't be using the brew version but direct from website? is brew version not just a scripted install of your own packages?

@scottnzuk can you get the SHA256 hash of the Stats.dmg that homebrew downloaded? I don't know where homebrew downloads it, but the terminal command to check the hash is `shasum -a 256 Stats.dmg`. It should be [`5a4362659ab63df1a9eb977cd981eea62733a234a4b4301ee29e168948f9c563`](https://github.com/Homebrew/homebrew-cask/commit/cf9aebb193a7726c4a06bfc1af239ab5e2877e95) for Stats 2.11.30

If it matches, go to your `/Applications/Stats.app` and press the space key for QuickLook, it should show the version number as Version 2.11.30

Then try the terminal command `spctl -avv /Applications/Stats.app`, it should show
```
/Applications/Stats.app: accepted
source=Notarized Developer ID
origin=Developer ID Application: Serhiy Mytrovtsiy (RP2S87B72W)
```

If all that matches, there is another issue

endlesslycurious, 2025-02-22:
[This is the source](https://github.com/Homebrew/homebrew-cask/blob/9983726f23d314d22f4ff3f0ae0503f3aecb0e2f/Casks/s/stats.rb) for the [HomeBrew cask recipe for stats](https://formulae.brew.sh/cask/stats), looking at the source it seems to be using the following Github release URL with `version "2.11.32"`
`url "https://github.com/exelban/stats/releases/download/v#{version}/Stats.dmg"`