exelban/stats #2397: Stats pops put after full screen

capemaster, 2025-02-16:
Every time, with every full screen.
https://youtu.be/_uoPyq-Gdho

exelban, 2025-02-16:
could you please explain that?
I don't get why opens? Do you klick somewhere or what?

capemaster, 2025-02-17:
Hi,
I posted a video concerning the behavior.

If you have Stats running and you watch a youtube video (for example) full screen, when you return to the normal view (not full screen), the stats app opens,
This happens all the time. With all apps.

I'm on a Mac, Sequoia.

capemaster, 2025-02-20:
> How it's possible that popup shows when you have no items in the menu bar?

You do not see them but stats is running. I think that the space is limited and sometimes some icons disappear. Mayb that is the case.
Anyway, I have tried to replicate the behavior, and it is erratic. I do not find a way to consistently replicate the bug.

(sorry I read the email only now)

exelban, 2025-02-20:
super strange, I can imagine that only if some 3d party tool calls that. Or maybe chrome full screen makes some shenanigans. (all times I use full screen mode in chrome I have strange behavior of macOS)

I will try to reproduce that, but nothing comes to mind about that.

capemaster, 2025-02-22:
Hi, I maybe found a way to reproduce... kinda.
Please note that I have activated the "combine modules" option.
If I press `esc` after activating the popup window, the popup changes status and shows me the CPU stats in full.

`esc` is the same button we press to exit from a full screen mode. Maybe some sort of keyboard shortcut overlap?

exelban, 2025-03-02:
sorry, I was not able to reproduce that.

closed 2025-03-02 by exelban