exelban/stats #2399: GPU Monitor switching between 100% and 0% frequently

pujux, 2025-02-18:
**Describe the bug**
For some reason my GPU monitor will seems to switch between 100% and 0% frequently. I do have a couple Chrome Tabs, Mail, Teams, VS Code and some other programs open (check screenshot below) but nothing that should utilize nor stress the GPU. 

[image]

Below is a screen recording showing my Activity Monitor sorted by GPU %. 

https://github.com/user-attachments/assets/1700f193-9602-4b57-b3f5-80589cb9b863

**Details:**
 - Model Name:	        MacBook Pro (Apple M4 Pro 14 core) 48GB RAM
 - macOS:               Sequoia 15.3 (24D60)
 - Application version: 2.11.29

exelban, 2025-02-22:
hi. It's for sure not a problem in the app itself. Most probably it's something with your drivers. Or maybe something specific to the m4 pro SoC. But first time hearing about that, so the first one is more probable. 

Maybe someone with m4 pro can confirm if they have the same problem?

closed 2025-03-02 by exelban

pujux, 2025-03-02:
Yeah I also think it could be an issue on my end more than on the apps side.

I just wanted to bring it up in case anyone else has had that problem and maybe even was able to fix it. I don't care about it too much because as soon as I get some real load on my GPU it feels like it shows correctly, so this is just happening in "idle".

anstadnik, 2025-06-06:
Hi, I'm having the same problem, m4 max. @pujux were you able to fix it? I see others having the same problem as well #2525, so IMHO it makes sense to reopen the issue.

pujux, 2025-10-31:
hey @anstadnik @indirect - I have not been able to fix it nor did I really try to be honest, I'm doing full stack work without any local AI models involved so I actually don't care much about GPU usage a lot 😅 

I was just used to having it there for quite some time so it was itching to see it switch between 100% and 0% all the time