exelban/stats #2401: Inconsistent reporting of disk free space

mike-ward, 2025-02-19:
In the screenshot note that the toolbar says 39% while while the dialog indicates 61%. 61% is the correct disk usage.

[image]

    Hardware Overview:

      Model Name: MacBook Air
      Model Identifier: Mac14,15
      Model Number: MQKW3LL/A
      Chip: Apple M2
      Total Number of Cores: 8 (4 performance and 4 efficiency)
      Memory: 8 GB
      System Firmware Version: 11881.81.4
      OS Loader Version: 11881.81.4
      Serial Number (system): HY3PVFK00X
      Hardware UUID: BC1D35E5-8372-5DF6-91DD-5702DF8B71F9
      Provisioning UDID: 00008112-001E09322108201E
      Activation Lock Status: Disabled

    System Software Overview:

      System Version: macOS 15.3.1 (24D70)
      Kernel Version: Darwin 24.3.0
      Boot Volume: Macintosh HD
      Boot Mode: Normal
      Secure Virtual Memory: Enabled
      System Integrity Protection: Enabled
      Time since boot: 6 hours, 21 minutes

    Disk:

      Filesystem        Size    Used   Avail Capacity iused ifree %iused  Mounted on
      /dev/disk3s1s1   228Gi    10Gi    62Gi    15%    412k  654M    0%   /
      devfs            199Ki   199Ki     0Bi   100%     688     0  100%   /dev
      /dev/disk3s6     228Gi    20Ki    62Gi     1%       0  654M    0%   /System/Volumes/VM
      /dev/disk3s2     228Gi   6.4Gi    62Gi    10%    1.2k  654M    0%   /System/Volumes/Preboot
      /dev/disk3s4     228Gi   3.7Mi    62Gi     1%      53  654M    0%   /System/Volumes/Update
      /dev/disk1s2     500Mi   6.0Mi   483Mi     2%       1  4.9M    0%   /System/Volumes/xarts
      /dev/disk1s1     500Mi   5.6Mi   483Mi     2%      32  4.9M    0%   /System/Volumes/iSCPreboot
      /dev/disk1s3     500Mi   808Ki   483Mi     1%      65  4.9M    0%   /System/Volumes/Hardware
      /dev/disk3s5     228Gi   147Gi    62Gi    71%    1.6M  654M    0%   /System/Volumes/Data 
      map auto_home      0Bi     0Bi     0Bi   100%       0     0     -   /System/Volumes/Data/home

closed 2025-02-19 by exelban

mike-ward, 2025-02-19:
Interesting. I searched the bugs for disk issues before submitting.

exelban, 2025-02-19:
hmm, what happens if you click on the text with free space?

mike-ward, 2025-02-20:
I was referring to which bug I duplicated. I didn't see one when I searched.  Probability didn't use the right search term and was curious what I missed.