exelban/stats #2404: Contribution requester popup every time the app runs?

schack-lindemann, 2025-02-20:
Every time I login and Stats startup I get a Contribution requester popup. This behaviour started with the latest version and seemingly can't be disabled. I hope this is unintentional. 

btw. I did contribute, both now and before.

dmnktoe, 2025-02-20:
same here! crashes since and can't reopen unfortunately.

exelban, 2025-02-21:
hi. Nope, it's not intentional. Looks like a bug. What exactly version of Stats do you use? And if the setup window shows as well?

dlesuisse, 2025-02-21:
Same problem here. Contribution request popup. I uninstalled the app, but after reinstalling, same behavior. Very irritating, but no crash. Version 2.11.31.

[image]

exelban, 2025-02-21:
not good, not good

exelban, 2025-02-21:
Im not sure I will find a time today to fix that, but will try to fix it till Sunday. If not, I will just disable this window till it will be fixed.

dlesuisse, 2025-02-21:
Thank you for your help! Have a nice day!

MrCyjaneK, 2025-02-21:
App was crashing on my end instantly after opening (i was able to see the contribution box for a brief 0.1s, multiple relaunches didn't fix the issue,

when I tried to go to terminal to get some logs this is what I got:
```
➜  ~ /Applications/Stats.app/Contents/MacOS/Stats 
2025-02-21 10:06:21 reader.swift:85 DBG [<CPU.LoadReader: 0x150e7b080>] Successfully initialize reader
2025-02-21 10:06:21 reader.swift:159 DBG [<CPU.ProcessReader: 0x600000b733c0>] Set update interval: 1 sec
2025-02-21 10:06:21 reader.swift:85 DBG [<CPU.ProcessReader: 0x600000b733c0>] Successfully initialize reader
2025-02-21 10:06:21 reader.swift:159 DBG [<CPU.AverageReader: 0x600000b73720>] Set update interval: 60 sec
2025-02-21 10:06:21 reader.swift:85 DBG [<CPU.AverageReader: 0x600000b73720>] Successfully initialize reader
2025-02-21 10:06:21 reader.swift:85 DBG [<CPU.TemperatureReader: 0x600000b737b0>] Successfully initialize reader
2025-02-21 10:06:21 reader.swift:85 DBG [<CPU.FrequencyReader: 0x600000e713f0>] Successfully initialize reader
2025-02-21 10:06:21 reader.swift:85 DBG [<GPU.InfoReader: 0x600000b678d0>] Successfully initialize reader
2025-02-21 10:06:21.764 Stats[62428:697506] NSPanel requires NSWindowStyleMaskUtilityWindow for a HUD window
2025-02-21 10:06:21 reader.swift:85 DBG [<RAM.UsageReader: 0x600000558800>] Successfully initialize reader
2025-02-21 10:06:21 reader.swift:159 DBG [<RAM.ProcessReader: 0x600000b76f40>] Set update interval: 1 sec
2025-02-21 10:06:21 reader.swift:85 DBG [<RAM.ProcessReader: 0x600000b76f40>] Successfully initialize reader
2025-02-21 10:06:21 reader.swift:85 DBG [<Disk.CapacityReader: 0x6000017fa300>] Successfully initialize reader
2025-02-21 10:06:21 reader.swift:159 DBG [<Disk.ActivityReader: 0x600001721280>] Set update interval: 1 sec
2025-02-21 10:06:21 reader.swift:85 DBG [<Disk.ActivityReader: 0x600001721280>] Successfully initialize reader
2025-02-21 10:06:21 reader.swift:159 DBG [<Disk.ProcessReader: 0x600000b7fa80>] Set update interval: 1 sec
2025-02-21 10:06:21 reader.swift:85 DBG [<Disk.ProcessReader: 0x600000b7fa80>] Successfully initialize reader
2025-02-21 10:06:21 reader.swift:85 DBG [<Sensors.SensorsReader: 0x600000255860>] Successfully initialize reader
2025-02-21 10:06:22 readers.swift:317 DBG [<Sensors.SensorsReader: 0x600000255860>] Found 2 fans
2025-02-21 10:06:22.645 Stats[62428:697506] NSPanel requires NSWindowStyleMaskUtilityWindow for a HUD window
2025-02-21 10:06:22 reader.swift:85 DBG [<Net.UsageReader: 0x140f132b0>] Successfully initialize reader
2025-02-21 10:06:22 reader.swift:85 DBG [<Net.ProcessReader: 0x600000b2c3f0>] Successfully initialize reader
2025-02-21 10:06:22 reader.swift:85 DBG [<Net.ConnectivityReader: 0x1170d7d20>] Successfully initialize reader
2025-02-21 10:06:22 reader.swift:85 DBG [<Battery.UsageReader: 0x117364540>] Successfully initialize reader
2025-02-21 10:06:22 reader.swift:85 DBG [<Battery.ProcessReader: 0x600001569680>] Successfully initialize reader
2025-02-21 10:06:22 reader.swift:85 DBG [<Bluetooth.DevicesReader: 0x600000861040>] Successfully initialize reader
2025-02-21 10:06:22 reader.swift:85 DBG [<Clock.ClockReader: 0x600001558b00>] Successfully initialize reader
Error Domain=NSCocoaErrorDomain Code=4099 "The connection to service named eu.exelban.Stats.SMC.Helper was invalidated: failed at lookup with error 3 - No such process." UserInfo={NSDebugDescription=The connection to service named eu.exelban.Stats.SMC.Helper was invalidated: failed at lookup with error 3 - No such process.}
2025-02-21 10:06:22 reader.swift:128 DBG [<Net.UsageReader: 0x140f132b0>] Set up update interval: 1 sec
2025-02-21 10:06:22 reader.swift:128 DBG [<Net.ConnectivityReader: 0x1170d7d20>] Set up update interval: 1 sec
2025-02-21 10:06:22 widget.swift:296 DBG [Network] widget speed enabled
2025-02-21 10:06:22 reader.swift:128 DBG [<Disk.CapacityReader: 0x6000017fa300>] Set up update interval: 1 sec
2025-02-21 10:06:22 reader.swift:128 DBG [<Disk.ActivityReader: 0x600001721280>] Set up update interval: 1 sec
2025-02-21 10:06:22 widget.swift:296 DBG [Disk] widget mini enabled
2025-02-21 10:06:22 reader.swift:128 DBG [<RAM.UsageReader: 0x600000558800>] Set up update interval: 1 sec
2025-02-21 10:06:22 widget.swift:296 DBG [RAM] widget mini enabled
2025-02-21 10:06:22 reader.swift:128 DBG [<GPU.InfoReader: 0x600000b678d0>] Set up update interval: 1 sec
2025-02-21 10:06:22 widget.swift:296 DBG [GPU] widget mini enabled
2025-02-21 10:06:22 reader.swift:128 DBG [<CPU.LoadReader: 0x150e7b080>] Set up update interval: 1 sec
2025-02-21 10:06:22 reader.swift:128 DBG [<CPU.FrequencyReader: 0x600000e713f0>] Set up update interval: 1 sec
2025-02-21 10:06:22 widget.swift:296 DBG [CPU] widget mini enabled
2025-02-21 10:06:22 helpers.swift:110 DBG Application update interval is 'At start'
2025-02-21 10:06:22 helpers.swift:151 DBG error updater.check(): last check was 7 minutes ago, stopping...
2025-02-21 10:06:22 AppDelegate.swift:85 INF Stats started in 0.0166 seconds
[1]    62428 trace trap  /Applications/Stats.app/Contents/MacOS/Stats
```

I was on stats `2.11.31` installed via brew, sadly/happily next runs (from terminal or from spotlight) worked just fine, but the popup is shown every time.

exelban, 2025-02-21:
@MrCyjaneK this issue does not relate to the crash. Please open another issue if the problem persists with crash report.

dayas-kr, 2025-02-21:
This is so annoying, pls stop this  Contribution requester popup!

exelban, 2025-02-21:
in case you does not want to wait till Sunday:

[Stats.dmg.zip](https://github.com/user-attachments/files/18908271/Stats.dmg.zip)

dlesuisse, 2025-02-21:
Thank you but same behavior here. I uninstalled the previous version correctly, rebooted and the same request appears... :-(

exelban, 2025-02-21:
yeap, but only for the first time. Try to restart the app and check if it pops again

dlesuisse, 2025-02-21:
You're right, it pops only the first time :-). Thank you and have a nice day!

exelban, 2025-02-21:
release with the fix is [available](https://github.com/exelban/stats/releases/tag/v2.11.32). Sorry for the bug, pretty stupid mistake(

I will leave this issue opened for a few days to prevent duplication.

dlesuisse, 2025-02-21:
Thanks for the great work!

schack-lindemann, 2025-02-22:
Thanks for the quick response. I found a temp. fix by resetting my preferences and re-enabling the modules I needed.

dmnktoe, 2025-02-24:
tysm. it works!

closed 2025-03-02 by exelban