exelban/stats #2406: Unwanted network connection

checktext00, 2025-02-21:
**Describe the bug**
Stats sometimes tries to connect to `api.mac-stats.com` on TCP port 443 even when "Check for updates" is set to "Never" and "Share anonymous telemetry" is disabled in Settings

**Details:**
 - macOS: 11
 - Application version: 2.11.30

exelban, 2025-02-21:
you need to disable public ip

checktext00, 2025-02-21:
> you need to disable public ip

it's already disabled, in Network > "Auto-refresh public IP address" is set to "Never" (which is already the default)

exelban, 2025-02-21:
hmm, it's only when Stats starts or with some interval when running?

checktext00, 2025-02-21:
I updated from 2.11.30 to 2.11.32 and the behavior now occurs on every launch of Stats instead of just when connecting to a VPN (I again reset all settings with "Check for updates" set to "Never" and "Share anonymous telemetry" disabled)

Also, a separate but related issue is that on a first-time launch Stats tries to connect to both `api.mac-stats.com` and `api.github.com` on the "Welcome to Stats" setup screen, before the user has made any choice about checking for updates or sharing telemetry data (tested on Stats 2.11.32)

closed 2025-03-02 by exelban