exelban/stats #2408: Can't open the app

simonescoca, 2025-02-21:
Don't know why the app fails at launch. I uninstalled and reinstalled it via homebrew, but the same issue occurs every time. Until yesterday, it was working perfectly.

**Details:**
 - Device: Macbook Air M3
 - macOS: 15.3.1
 - Application version: latest available

exelban, 2025-02-21:
please share the crash report

simonescoca, 2025-02-21:
[Stats_CrashReport.txt](https://github.com/user-attachments/files/18913299/Stats_CrashReport.txt)

exelban, 2025-02-21:
[Stats.dmg.zip](https://github.com/user-attachments/files/18913396/Stats.dmg.zip)

thx, could you please try this build?

simonescoca, 2025-02-21:
It works, thank you 🙏🏼

exelban, 2025-02-21:
new release with fix is available

closed 2025-02-21 by exelban

saqibshafin, 2025-02-27:
> new release with fix is available

Thank you so much, brother.
Keep up the great work. It means a lot to us.