exelban/stats #2417: Random crashes

KingfuChan, 2025-02-25:
Random crashes keeps happening. Sometimes I get to restart the app for multiple times in a single day. I've already searched for any opened issue, but didn't find any similar topics.

My system info is as follows:

[image]

No idea whether there's any log file to share. Here's my config.

[Stats.plist.txt](https://github.com/user-attachments/files/18968760/Stats.plist.txt)

exelban, 2025-02-25:
please share the crash report

KingfuChan, 2025-02-26:
That would be the tricky part. Stats would just disappear from my menu bar without any window prompting a crash of the app. Or is there anywhere else I can find a report?

exelban, 2025-02-26:
It could be found in Applications -> Utilities -> Console. Find the Crash Reports tab. There you must see a row called Stats.

KingfuChan, 2025-02-26:
> It could be found in Applications -> Utilities -> Console. Find the Crash Reports tab. There you must see a row called Stats.

Found them:

[Stats-2025-02-20-113722.txt](https://github.com/user-attachments/files/18979829/Stats-2025-02-20-113722.txt)
[Stats-2025-02-25-180112.txt](https://github.com/user-attachments/files/18979830/Stats-2025-02-25-180112.txt)
[Stats-2025-02-25-215017.txt](https://github.com/user-attachments/files/18979832/Stats-2025-02-25-215017.txt)
[Stats-2025-02-25-233828.txt](https://github.com/user-attachments/files/18979831/Stats-2025-02-25-233828.txt)

exelban, 2025-02-26:
please update to the latest version of Stats

KingfuChan, 2025-02-26:
> please update to the latest version of Stats

Already on 2.11.32 when these crashes happened.

exelban, 2025-02-26:
could you please try to disable network module and check if the crash will continue?

exelban, 2025-02-27:
@bymaer update to the latest version of Stats

delef, 2025-02-28:
@exelban Yeah, I too am getting a crash on startup after the latest update

ssi-anik, 2025-02-28:
All of a sudden, the metrics are no longer showing in my menu bar for the last couple of days. [Looking in the console](https://github.com/exelban/stats/issues/2417#issuecomment-2684227856), I can see that the app crashes sometimes.

exelban, 2025-02-28:
please ensure that your are using the latest version of Stats. It has a fix that prevents crash

KingfuChan, 2025-03-02:
I just got another crash with lates 2.11.33. Network module is enabled. I've tried various network module config but all of them would crash the app. Disabling network module seems fine. But network module is the indispensable feature for me.

[Stats-2025-03-02-230435.ips.txt](https://github.com/user-attachments/files/19043605/Stats-2025-03-02-230435.ips.txt)

exelban, 2025-03-02:
please try this build:

[Stats.dmg.zip](https://github.com/user-attachments/files/19044141/Stats.dmg.zip)

exelban, 2025-03-04:
@KingfuChan any updates?

KingfuChan, 2025-03-04:
> [@KingfuChan](https://github.com/KingfuChan) any updates?

The test build is working so far so good.

I noticed a change in the public IP feature and I disabled both toggle. Though I don't know if it's a factor since I kept in on before.

alexeypegov, 2025-03-05:
Started to crash today right at the startup (2.11.33).

[Stats-2025-03-05-102129.txt](https://github.com/user-attachments/files/19088177/Stats-2025-03-05-102129.txt)
[Stats-2025-03-05-121246.txt](https://github.com/user-attachments/files/19088185/Stats-2025-03-05-121246.txt)

exelban, 2025-03-05:
@alexeypegov you are running v2.11.31

alexeypegov, 2025-03-05:
@exelban ah, got it, that's interesting, since I have installed it via homebrew and it autoupdated itself, and homebrew tells me the latest version while it might failed to autoupdate since it crashes at the start. I have forced it to update and everything is fine now. Thanks!

2p31-1, 2025-03-06:
It started to crash immediately when it opened

vvatikiotis, 2025-03-08:
Same here. v2.11.33. MacOS 14.7.4, mac mini M1

[Stats-2025-03-08-151926.txt](https://github.com/user-attachments/files/19143819/Stats-2025-03-08-151926.txt)

Thanks for the app 😄

exelban, 2025-03-08:
@vvatikiotis why are you write `v2.11.33` if you are using `2.11.26`?

Please at least check what are you posting.

closed 2025-03-09 by exelban

vvatikiotis, 2025-03-09:
> [@vvatikiotis](https://github.com/vvatikiotis) why are you write `v2.11.33` if you are using `2.11.26`?
> 
> Please at least check what are you posting.

Oops, my mistake! I have installed Stats via brew *and* as a standalone app and I checked the brew version (2.11.33) while I was running the app. Honest mistake, sorry for the confusion!

KingfuChan, 2025-03-11:
[Stats-2025-03-11-232920.ips.txt](https://github.com/user-attachments/files/19188291/Stats-2025-03-11-232920.ips.txt)

Just now, another crash happened with the latest 2.11.34 release.

KingfuChan, 2025-03-21:
[Stats-2025-03-21-190355.ips.txt](https://github.com/user-attachments/files/19387971/Stats-2025-03-21-190355.ips.txt)

Here's another crash with 2.11.35

exelban, 2025-03-21:
@KingfuChan please try this build:

[Stats.dmg.zip](https://github.com/user-attachments/files/19388316/Stats.dmg.zip)

KingfuChan, 2025-03-21:
> [@KingfuChan](https://github.com/KingfuChan) please try this build:
> 
> [Stats.dmg.zip](https://github.com/user-attachments/files/19388316/Stats.dmg.zip)

Downloaded it just now. When I first run it, it shows a remote user login dialog. What's this for? I cancelled it though and Stats runs normally.

exelban, 2025-03-21:
yeap, just cancel that

KingfuChan, 2025-04-11:
[Stats-2025-04-11-195053.ips.txt](https://github.com/user-attachments/files/19704924/Stats-2025-04-11-195053.ips.txt)

Another crash just happen when I'm seeing Average CPU temp is going up to 60°C.