exelban/stats #2421: Battery detail doesn't show the time remaining

rantoniuk, 2025-02-27:
**Describe the bug**

The battery _details_ widget only shows percentage remaining while it should show also the remaining battery time. Upon clicking, it shows correctly the remaining time in the dropdown window.

**Details:**
 - Device: Macbook Pro 2019
 - macOS: 15.13.1
 - Application version: 2.11.32

exelban, 2025-02-27:
sorry but I do not understand the problem. Could you please give more details?

rantoniuk, 2025-02-27:
There are two battery status icons: "Mini" and "Battery details widget". They should display different things, but they don't. The _details_ widget lacks the remaining time.

[image]

exelban, 2025-02-27:
and you have enable that options?

exelban, 2025-02-27:
[image]

everything works fine, so please check your settings

closed 2025-02-27 by exelban