exelban/stats #2422: Feedback and problems

Volodymyr-13, 2025-02-28:
1. Confusing naming. "macOS system monitor in your menu bar" so for **menu bar**, so Menu Bar. What is widgets? macOS widgets? Also, Modules should be **second** tab, so Menu Bar tab(aka currently wrong name Widgets) should be **first** tab

[image]

2. Can't resize window for this option, where I can see a loooot of sensors, so allow me to resize vertically at least!

[image]

3. Why so much empty space here?

[image]

4. I want to have a control of **each state** for pressure/utilization, so like normal/medium/hard or whatever that names states are. So I want to set each specific color for each state.

[image]

5. I'm using Mac fans app for fans and I want 2 lines option here too, looks really good and compact

[image]

6. I'm using Wattagio and I want to be able to to add the same icons into menu bar, so the precise battery % and wattage.

[image]

7.  For RAM I want to be able to see a Swap size in the menu bar icon. 

**Details:**
 - Device: Macbook Pro 14-inch, 2021
 - macOS: 15.3.1 (24D70)
 - Application version: 2.11.32

exelban, 2025-02-28:
1 - no comments
2 - yes, the window cannot be resized, it will cause a lot of problems related to the different window sizes. I have no time to take care of that and don't see any benefits of that
3 - see closed issues
4 - ok, PR is welcome
5 - so just do what you want using stack widget
6 - battery wattage is available in sensors module
7 - again see closed issues

closed 2025-02-28 by exelban

Volodymyr-13, 2025-02-28:
Uninstalled...